INTI-CMNB/KiBot #401: Issue generating STEP with WSL

jurasamo, 2023-03-09:
Hello!,

  First of all thanks for this great project.

 I have been running kibot on WSL with kicad 6 without issues, but in a new computer with kicad 6.0.11 if I run kibot with my user i get this error generating the STEP.

- 'STEP 3D model' (3D) [step]
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.0', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 403, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 515, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 506, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 419, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 192, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_step.py", line 84, in run
    board_name = self.filter_components()
  File "/usr/lib/python3/dist-packages/kibot/out_base_3d.py", line 282, in filter_components
    ret = self.save_tmp_board()
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 870, in save_tmp_board
    GS.copy_project(fname)
  File "/usr/lib/python3/dist-packages/kibot/gs.py", line 388, in copy_project
    copy2(pro_name, pro_copy)
  File "/usr/lib/python3.10/shutil.py", line 435, in copy2
    copystat(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.10/shutil.py", line 374, in copystat
    lookup("utime")(dst, ns=(st.st_atime_ns, st.st_mtime_ns),
PermissionError: [Errno 1] Operation not permitted

If I run it with sudo it works well

Thanks for your help !,

set-soft, 2023-03-09:
Hi @jurasamo !

It looks like your regular user doesn't have access to all the project files. Or to the temporal dir. The error happens when KiBot is trying to copy the current .kicad_pro file to a temporal dir.

jurasamo, 2023-03-09:
Hi Salvador!

  Thank you for your quick answer.

 I see that is something related to permissions, but I can't find the problem. Kibot creates a temp files in the project directory (which I have given 777 permissions), does it create more temp files in other directories ?

set-soft, 2023-03-10:
Hi @jurasamo !
All temporal files are created using the Python API. Files alone are created at the same dir where the project is. But when we need a whole folder the system temporal directory should be used.

What about the *.kicad_pro files? Are they read-write?

set-soft, 2023-03-10:
Hi @jurasamo !
Are you sure that the Linux version of `kicad2step` is in the path (before the Window copy) because you should never see the window of the Linux version, KiAuto runs `kicad2step` in a virtual X server. But I know WSL can mix binaries (Linux calling Windows and vice versa)

set-soft, 2023-03-18:
Hi @jurasamo !
Any news about it?

jurasamo, 2023-03-20:
Hi @set-soft ,

  Sorry for my late answer, I had to change my focus. I'm back at work with this issue.

  We have doubled checked the PATH, and the path is right, actually what happens is that the Kicad2Step works under X11, not the Windows native.

  I have ran the script with sudo and it returns: ERROR:/usr/bin/kicad2step_do returned 245 (kibot - kiplot.py:201)

   Regards !

set-soft, 2023-03-21:
Hi @jurasamo !

Avoid using sudo to run the script, you'll have problems when trying to run it as a regular user and some files are now owned by root.
If kicad2step is generating a time-out then run KiBot using maximum debug info (-vvvv) to detect where is stuck.
Also: if the kicad2step window is visible then something is really wrong, the debug log could help to understand what.

set-soft, 2023-03-22:
Hola @jurasamo !

The window never opens, and hence the video is empty. You mentioned you see the window, if this is the case then something is really wrong in this setup. Most probably related to Xvfb tool. The window should be connected to the virtual X server, not the real X server. So it can't be visible, just drawing to a frame buffer in memory. The virtual X server seems to start (*Current DISPLAY is :1579592939*), the recorder seems to be recording the virtual X server (blank screen). But kicad2step doesn't create its window, and from what you say is connecting to the real X server.

Can you make a screen capture of the window?

Saludos, Salvador

set-soft, 2023-04-15:
Hi @jurasamo !
Any news?

set-soft, 2023-05-12:
Closing it because inactive, please re-open if you have more information

closed 2023-05-12 by set-soft