INTI-CMNB/KiBot #402: [BUG] AttributeError: 'PresentBoards' object has no attribute 'fill_empty_values'

DusterTheFirst, 2023-03-10:
Fill as much as you can, remove the rest

**Describe the bug**
When running `diff` on a schematic, the following error occurs:

```
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.0', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 403, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 515, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 504, in _generate_outputs
    if config_output(out, dont_stop=dont_stop):
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 382, in config_output
    out.config(None)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 148, in config
    super().config(parent)
  File "/usr/lib/python3/dist-packages/kibot/optionable.py", line 232, in config
    self._perform_config_mapping()
  File "/usr/lib/python3/dist-packages/kibot/optionable.py", line 196, in _perform_config_mapping
    v.config(self)
  File "/usr/lib/python3/dist-packages/kibot/out_kikit_present.py", line 319, in config
    a_board.fill_empty_values()
AttributeError: 'PresentBoards' object has no attribute 'fill_empty_values'
```

which points to
https://github.com/INTI-CMNB/KiBot/blob/cc2272e7028e43a2f880d06fc7c6467bfd7869fb/kibot/out_kikit_present.py#L319

which is calling a non-existent function non a_board, 

**To Reproduce**
With the following config:
```yml
outputs:
   - name: Changes
      comment: PCB and Schematic diff
      type: diff
      dir: diff
      layers: all
```

run `kibot -c config.kibot.yml -b avionics/avionics.kicad_pcb -e avionics/avionics.kicad_sch -d kibot` in the root of https://github.com/aeroteameindhoven/electronics/blob/main

**Expected behavior**
A crash does not occur

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad6_auto_full:latest`

set-soft, 2023-03-10:
Hi @DusterTheFirst !
I'll take a look at this the next week. In the mid time please try adding at least one board, like this:

```yaml
outputs:
    - name: Project Presentation
      comment: Presentation of the project
      type: kikit_present
      dir: present
      options:
        description: "README.md" # FIXME:
        boards:
          mode: local
```

DusterTheFirst, 2023-03-10:
That does remove the crash, it would be nice to have a better error message though.

set-soft, 2023-03-18:
The above patch allows skipping the `boards` option.

commit 2d398904 (2023-03-18): [KiKit present][Fixed] Crash when no boards specified

Fixes #402
closed 2023-03-18 by set-soft