INTI-CMNB/KiBot #403: [BUG] Hierarchical Schematic file name diff problem

DusterTheFirst, 2023-03-10:
**Describe the bug**
When making a diff of a sheet with hierarchical sub-sheets, their names will be shown as different even if it is the same old and new file

**To Reproduce**
Make a commit with changes not to any of the schematics, run kibot with the following config
```
outputs:
    - name: Schematic Changes
      comment: Schematic diff
      type: diff
      dir: sch
      options:
          pcb: false
          only_different: true
          old: HEAD~
```

**Expected behavior**
The diff should show no changes

**Screenshots**
[avionics-diff_sch.pdf](https://github.com/INTI-CMNB/KiBot/files/10942810/avionics-diff_sch.pdf)

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? `kicad6_auto_full:latest`

DusterTheFirst, 2023-03-10:
That did fix the problem. I am not using any variants (there is no variants section of my config) so I did not think that kibot would modify my sheet.

I do also want to applaud your work on this tool. It has been a great tool in my workflow.

set-soft, 2023-03-10:
> I am not using any variants (there is no variants section of my config) s

Sometimes subtle actions forces KiBot to create a variant of the file. You can see it enabling debug (i.e. -vvv)

closed 2023-03-10 by set-soft

set-soft, 2023-04-28:
Hi @DusterTheFirst !

The current `dev` version (1.6.3) should fix this issue. Now we avoid expanding the schematic hierarchy unless this is really necesary, i.e. when you force two instances of the same component from the same shared sheet to be different (even for the value, supported by KiCad 6, but not by KiCad 7)