INTI-CMNB/KiBot #404: [HOW] What is the intended way to use kibot with gitlab CI and a multipages projekt

MarkusSBS, 2023-03-15:
**What do you want to achieve?**
I have a projekt with multible hierarchical pages (33 pages) and i want to use it with kibot to generate Produktion data and Documentation. It has worked for the example projekt with just 1 page and the config of the gitlab example: https://gitlab.com/set-soft/kicad-ci-test

**Do you have some PCB/Schematic to use as example?**
My CI config: 

```
erc:
  stage: run_erc
  tags:
    - KiBot
  script:
    - "[ -f *.kicad_sch ] && kibot -d ERC_DRC -s update_xml,run_drc -i"
  rules:
    - if: $CI_COMMIT_TAG =~ /^(preview|prerelease|release)_([0-9]{2}.[0-9]{4}.[0-9]{2})_([a-z,A-Z][0-9])/
    - changes:
      - "**.kicad_sch"
      - ".gitlab-ci.yml"
      when: always
  artifacts:
    expire_in: 1 week
    when: always
    paths:
      - ERC_DRC/

drc:
  stage: run_drc
  tags:
    - KiBot
  script:
    - "[ -f *.kicad_pcb ] && kibot -d ERC_DRC -s update_xml,run_erc -i"
  rules:
    - if: $CI_COMMIT_TAG =~ /^(preview|prerelease|release)_([0-9]{2}.[0-9]{4}.[0-9]{2})_([a-z,A-Z][0-9])/
    - changes:
      - "**.kicad_pcb"
      - ".gitlab-ci.yml"
      when: always
  artifacts:
    expire_in: 1 week
    when: always
    paths:
      - ERC_DRC/

docu_outputs:
  stage: gen_fab
  tags:
    - KiBot
  script:
    - "[ -f *.kicad_sch ] && kibot  -d Docu -s run_drc,run_erc print_sch print_pcb_pdf report_full report_history report_pcb_svg interactive_bom"
  rules:
    - if: $CI_COMMIT_TAG =~ /^(preview|prerelease|release)_([0-9]{2}.[0-9]{4}.[0-9]{2})_([a-z,A-Z][0-9])/
  needs: 
    - job: erc
    - job: drc
  artifacts:
    expire_in: 1 week
    when: always
    paths:
      - Docu/

```
** problem **

With this config all kicad_sch related JOBs fail with 

```
$ [ -f *.kicad_sch ] && kibot -d ERC_DRC -s update_xml,run_drc -i
/bin/bash: line 129: [: too many arguments
```

If i changed it to:

```
 [ -f *.kicad_pro] && kibot -d ERC_DRC -s update_xml,run_drc -i
```

i get the warning
```
 WARNING:(W044) More than one SCH file found in `.`.
  Using ./EMOB2.kicad_sch if you want to use another use -e option. (kibot - kiplot.py:753)
```
I am not sure of the consequences of this config.

what i noticed: the ERC kibot run only generates only 63errors, the ERC run in KiCAD  over 800 (its a new projekt work in progress)

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? `kicad_auto:dev_k6`,

set-soft, 2023-03-15:
> maybe a mechanisem would be nice to have a generic Approach by the Projekt file?

The check in the example looks for at least one schematic, this can't be replaced by checking for a project. You can have a project and no schematic. In particular in a repo. KiCad 6 doesn't create an empty schematic when you create a project.

> For a standart KIcad projekt the name of the kicad_pro file should be the same as the toplevel sheet and the main PCB.
The kicad_pro should be also uniqe in the folder for most projekts

And this is why kibot tries to guess the schematic looking for a project. But I recommend doing things explicit for a CI/CD environment. For your interactive command line you can just run *kibot* 99 % of the time and it will do what you want. What's the name of your project file?

MarkusSBS, 2023-03-15:
> The check in the example looks for at least one schematic, this can't be replaced by checking for a project. You can have a project and no schematic. In particular in a repo. KiCad 6 doesn't create an empty schematic when you create a project.

For a repo it can be true but a realy specific Konstellation. For KiCad 6 I  tested it: It is imposible to creat a new projekt without a  empy SCH or PCB.(on Windows)

> What's the name of your project file?

Projekt file name is EMOB2.kicad_pro

`[ -f *.kicad_pcb ] &&` 

And 

`[ -f *.kicad_sch ] &&`

only checks if a file Exits?  sorry for not the fittest in bash syntax 

For a job like the following how generates a big chunck of the documention is it critical to specify only the SCH or BRD with the -e option:

` - "[ -f *.kicad_sch ] && kibot  -d Docu -s run_drc,run_erc print_sch print_pcb_pdf report_full report_history report_pcb_svg interactive_bom"`

it contains element for the pcb and the schematic

set-soft, 2023-03-15:
> For a repo it can be true but a realy specific Konstellation

But the example you show is exactly for a repo! As I told you: for a shell just run *kibot*, if it does a wrong guess you can just try again, for a CI/CD if something is wrong you'll waste 15 or more minutes, not 2 seconds.

> Projekt file name is EMOB2.kicad_pro

Then kibot did a correct guess choosing *./EMOB2.kicad_sch*

> [ -f *.kicad_pcb ] && 
> only checks if a file Exits? sorry for not the fittest in bash syntax

Yes, you can ask it to "chatGPT", like this `What does " [ -f *.kicad_pcb ] && " mean for a shell script?` the answer is correct:  (it also understands german)

<-- chatGPT response
In a shell script, [ -f *.kicad_pcb ] is a conditional statement that checks if there is a file in the current directory with a name that matches the pattern *.kicad_pcb.

The && operator is a logical AND operator. It means that the command following && will only be executed if the command preceding && returns a success status (i.e., exit code 0).

Putting it all together, [ -f *.kicad_pcb ] && means that if there is a file in the current directory with a name that matches the pattern *.kicad_pcb, then the command following && will be executed.

For example, a possible shell script could be:

```
#!/bin/bash
if [ -f *.kicad_pcb ]; then
    echo "Found a Kicad PCB file!"
fi
```

In this case, if there is at least one file with a name that matches the pattern *.kicad_pcb in the current directory, the script will print "Found a Kicad PCB file!" to the console.
<-- end of chatGPT response

Is funny the example generated by the AI, correctly inferring this detects a KiCad PCB file.

> For a job like the following which generates a big chunck of the documention is it critical to specify only the SCH or BRD with the -e option:

Yes, most of the time. I'll suggest passing the name of the PCB using *-b*. This will be the same as the project, and will avoid any confusion if you accidentally put two projects on the same folder.

set-soft, 2023-03-18:
I changed the test for one that is more robust (and less readable): `[ -n '$(find . -maxdepth 1 -name *.kicad_pcb -print -quit)' ] &&`

closed 2023-03-18 by set-soft