INTI-CMNB/KiBot #407: [BUG] pcb_print output depends on layer visibility in .kicad_prl file

richardbarlow, 2023-03-24:
**Describe the bug**
The output of the `pcb_print` output type is dependent on the layer visibility set in the .kicad_prl file. When using the scaling option layers that are set as visible in the associated .kicad_prl file affect the positioning of elements in the output PDF, even when those layers are not included in the 'layers' list in the KiBot output config.

I do not commit the .kicad_prl file to version control (since these are the local settings specific to the user). Therefore, the behaviour of KiBot locally and in CI differs. In addition, this means that even if layers are not listed in the KiBot output config, they impact the PDF output.

**To Reproduce**
Run `kibot -b kibot_pcb_print_prl_effect.kicad_pcb -c ci.kibot.yaml -d output` with the following files:
[kibot_pcb_print_prl_effect.zip](https://github.com/INTI-CMNB/KiBot/files/11063746/kibot_pcb_print_prl_effect.zip)

Try opening the layout in pcbnew and hiding F.Cu, Edge.Cuts and User.Eco1 in different combinations and observe the effect on the output.

**Expected behavior**
Only layers listed in the KiBot config for the `pcb_print` output type affect the PDF output. When scaling (manual or auto) the listed layers should be centred in the title block and not influenced by unlisted layers (regardless of if the .kicad_prl file has them visible or not).

**Screenshots**
With F.Cu, Edge.Cuts and User.Eco1 all visible in .kicad_prl:
[image]

With only F.Cu visible in .kicad_prl:
[image]

With only User.Eco1 visible in .kicad_prl:
[image]

**Environment (please complete the following information):**
KiBot version 1.6.0 run locally (Fedora 35) and v2_k6_1_6_0 in GitHub Actions/

set-soft, 2023-03-27:
Hi @richardbarlow !
Never noticed it, thanks for reporting.
Is ridiculous that the Python API loads the PRL file.
Now the code enables only the layers used for the current page.

I think a similar problem must affect all the plotting outputs, I'll check this tomorrow.

closed 2023-03-27 by set-soft

richardbarlow, 2023-03-30:
This works beautifully. Thanks @set-soft!

commit ac81d18e (2023-03-27): [PCB Print][Fixed] Interference with PRL and scaling

- The visible layers affected the position of the scaled image.
  This list is defined in the .kicad_prl file and the Python API
  loads it.

Fixes #407