INTI-CMNB/KiBot #408: [BUG] Issue downloading datasheets

KenwoodFox, 2023-03-25:
Looks like my run fails at about the time its trying to download a datasheet, I'm not sure why its doing this. Id like to just disable it if i can, i don't need it to try and download any of my datasheets

```
 - 'Download the datasheets' (basic_download_datasheets) [download_datasheets]
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/kibot/__main__.py", line 363, in main
    generate_examples(args.start, args.dry, args.type)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 1030, in generate_examples
    generate_targets(c)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 966, in generate_targets
    generate_outputs(outputs, [], False, None, False, False, dont_stop=True)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 521, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 512, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.9/dist-packages/kibot/kiplot.py", line 423, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_download_datasheets.py", line 167, in run
    self.options.run(output_dir)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_download_datasheets.py", line 131, in run
    name = self.download(c, ds, output_dir, name, known)
  File "/usr/local/lib/python3.9/dist-packages/kibot/out_download_datasheets.py", line 68, in download
    r = requests.get(ds, allow_redirects=True, headers={'User-Agent': USER_AGENT}, timeout=20)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 677, in send
    history = [resp for resp in gen]
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 677, in <listcomp>
    history = [resp for resp in gen]
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 166, in resolve_redirects
    raise TooManyRedirects('Exceeded {} redirects.'.format(self.max_redirects), response=resp)
requests.exceptions.TooManyRedirects: Exceeded 30 redirects.
```

Currently failing with [this project here](https://github.com/KenwoodFox/MegaMouse/pull/1)

This is a CI/CD workflow.

Heres what it looks like if i run it from the command line locally.
```
kibot -e MegaMouse/MegaMouse.kicad_sch -c config.kibot.yaml -d _output -b MegaMouse/MegaMouse.kicad_pcb
- Updating BoM in XML format
- 'Schematics and media for use in docs' (Images) [pdf_sch_print]
- '3D Models' (3D Models) [step]
WARNING:(W048) Failed to download `https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Connector_USB.3dshapes/USB_C_Receptacle_XKB_U262-16XN-4BVC11.wrl` (kibot - out_base_3d.py:115)
WARNING:(W047) Missing 3D model for J601: `/usr/share/kicad/3dmodels/Connector_USB.3dshapes/USB_C_Receptacle_XKB_U262-16XN-4BVC11.wrl` (kibot - out_base_3d.py:252)
- '3D Renders for docs!' (3D Renders) [render_3d]
WARNING:(W058) Old file format detected, please convert it to KiCad 6 if experimenting problems (pcbnew_do - interposer.py:656) (kibot - kiplot.py:134)
- 'Gerbers compatible with PCBWay' (PCBWay_gerbers) [gerber]
- 'Drill files compatible with PCBWay' (PCBWay_drill) [excellon]
- 'Gerbers compatible with JLCPCB' (JLCPCB_gerbers) [gerber]
- 'Drill files compatible with JLCPCB' (JLCPCB_drill) [excellon]
- 'Pick and place file, JLCPCB style' (JLCPCB_position) [position]
WARNING:(W013) `REF**` component in board, but not in schematic (kibot - kiplot.py:302)
WARNING:(W013) `mouse-bite-5mm-slot` component in board, but not in schematic (kibot - kiplot.py:302)
WARNING:(W062) Invalid column name `LCSC#`. Valid columns are ['Row', 'Description', 'Part', 'Part Lib', 'References', 'Value', 'Footprint', 'Footprint Lib', 'Quantity Per PCB', 'Build Quantity', 'Status', 'Datasheet', 'Sheetpath', 'Source BoM', 'Reference', 'Footprint X', 'Footprint Y', 'Footprint Rot', 'Footprint Side', 'Footprint Type', 'Footprint Populate', 'Footprint X-Size', 'Footprint Y-Size']. (kibot - out_bom.py:667)
- 'BoM for JLCPCB' (JLCPCB_bom) [bom]
- 'ZIP file for PCBWay' (PCBWay) [compress]
- 'ZIP file for JLCPCB' (JLCPCB) [compress]
Found 6 unique warning/s (24 total)
```

Thanks for taking a look at this!

KenwoodFox, 2023-03-27:
Heres a link to the relevant actions section: https://github.com/KenwoodFox/MegaMouse/blob/5d39779a7c9d6032e89dd3b7008820f2fa78dcb1/.github/workflows/hardware_workflow.yml#L15-L49

closed 2023-03-27 by set-soft

KenwoodFox, 2023-03-27:
Actually deleting the datasheet was the first thing i tried.
[image]
Still no-go.

set-soft, 2023-03-27:
Hi @KenwoodFox !

First a note: the patch isn't yet deployed, GitHub is experimenting some serious degradation.

> Actually deleting the datasheet was the first thing i tried. [image] Still no-go.

I guess you are removing the wrong URL.

> Is there a way to get more verbose debugging? 

Us `-v` or even `-vvvv`, as documented.

> if its just a warning why does it crash? 

Again: the last patch isn't yet deployed. The regression tests failed because the Ghostscript people released v10 some days ago, but didn't include Linux binaries. The workaround for this isn't yet in the repo because GitHub is rejecting pushes.

> And why does it work locally but not with the github action docker image?

The network offered by GitHub is filtered and shaped. Without even knowing the faulty URL (which will be displayed if you used debug information, even on fail/crash) is hard to know.

KenwoodFox, 2023-03-27:
I didn't know that there was a patch, or that it was undeployed, did you say that earlier?
Thanks ill try with more verbose maybe its a different device? Should i just wait for the patch to apply though?

set-soft, 2023-03-27:
> I didn't know that there was a patch, or that it was undeployed, did you say that earlier? 

If you take a look at the history, you'll see:

[set-soft](/set-soft) added a commit that referenced this issue [ Mar 27, 2023 ](#ref-commit-f7d52f3)
[ [image] ](/set-soft)
 [[Datasheet dowload] Catch requests.exceptions.TooManyRedirects](/INTI-CMNB/KiBot/commit/f7d52f3d093ab41aba4f6a36533872339c581dee)  …

Is just before my reply saying *Now this should be just a warning.*.  

> Thanks ill try with more verbose maybe its a different device? 

Most probably. I added a second path to display the faulty reference.

> Should i just wait for the patch to apply though?

Yes, you must wait until the *actions* pass the tests.

KenwoodFox, 2023-03-27:
Oh duh, ok i see the patch now at f7d52f3d093ab41aba4f6a36533872339c581dee, sorry idk how i missed that it just wasn't commented on. Awesome thanks so much!

set-soft, 2023-03-27:
Now is deployed. GitHub is working normally again.

commit f7d52f3d (2023-03-27): [Datasheet dowload] Catch requests.exceptions.TooManyRedirects

Closes #408