INTI-CMNB/KiBot #410: [BUG] Parity Check Generating Errors not seen in KiCad

jtmullen, 2023-03-29:
**Describe the bug**
I am running kibot in docker in Github Actions. I am simply doing ERC/DRC/Parity Check in this workflow. Locally I get no parity issues in the DRC. Additionally, when I put intentional parity issues those issues are reproduced exactly between local and kibot in addition to the erroneous errors. 

Specifically, with a simple test pcb with a non-sense circuit connecting a diode, resistor, and capacitor I get this output: 

```md
WARNING:(W058) Missing KiCad main config file /github/home/.config/kicad/6.0/kicad_common.json (eeschema_do - misc.py:198) (kibot - kiplot.py:134)
ERROR:PCB net code 1 name mismatch (Net-(C1-Pad1) vs +3V3) (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 1 extra connection/s: C1 pin 1,R1 pin 2 (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 1 missing connection/s: D1 pin 2 (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 3 name mismatch (Net-(D1-Pad1) vs Net-(C1-Pad1)) (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 3 extra connection/s: D1 pin 1,R1 pin 1 (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 3 missing connection/s: C1 pin 1,R1 pin 2 (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 4 name mismatch (+3V3 vs Net-(D1-Pad1)) (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 4 extra connection/s: D1 pin 2 (kibot.pre_update_xml - pre_update_xml.py:169)
ERROR:PCB net code 4 missing connection/s: D1 pin 1,R1 pin 1 (kibot.pre_update_xml - pre_update_xml.py:169)
```

Especially confusing to me about this is how some of these are contradictory, such as `C1 pin 1,R1 pin 2` being listed as both a missing and extra connection. 

**To Reproduce**
I am simply running pre-flight checks with this config: 

```yml
kibot:
  version: 1
  
global:
  drc_exclusions_workaround: true

## Tasks to do before output
preflight:
  update_xml:  ## Ensure PCB and Schematic match
    check_pcb_parity: true
    enabled: true
  run_erc: true
  run_drc: true
  check_zone_fills: true
  ignore_unconnected: false
```

**Expected behavior**
The output should be the same errors as KiCad produces. 

**Environment (please complete the following information):**
Local:  KiCad 6.0.7 on Windows 11
GitHub Actions:  in `setsoft/kicad_auto:ki6`

jtmullen, 2023-03-30:
Hi @set-soft, sorry this was in a private repo. I've put a minimal example that reproduces the issue here: https://github.com/jtmullen/kibot-test/pull/1

Same result on the GH Actions run here: https://github.com/jtmullen/kibot-test/actions/runs/4567596718?pr=1

set-soft, 2023-03-31:
Hi @jtmullen !

This is some KiCad inconsistency, take a look at the netlist, it says:

```xml
  <nets>
    <net code="1" name="+3V3">
      <node ref="D1" pin="2" pinfunction="A" pintype="passive"/>
    </net>
    <net code="2" name="GND">
      <node ref="C1" pin="2" pintype="passive"/>
      <node ref="D1" pin="3" pinfunction="K" pintype="passive"/>
    </net>
    <net code="3" name="Net-(C1-Pad1)">
      <node ref="C1" pin="1" pintype="passive"/>
      <node ref="R1" pin="2" pintype="passive"/>
    </net>
    <net code="4" name="Net-(D1-Pad1)">
      <node ref="D1" pin="1" pinfunction="A" pintype="passive"/>
      <node ref="R1" pin="1" pintype="passive"/>
    </net>
  </nets>
```

Now take a look at the PCB, it says:

```lisp
  (net 0 "")
  (net 1 "Net-(C1-Pad1)")
  (net 2 "GND")
  (net 3 "Net-(D1-Pad1)")
  (net 4 "+3V3")
```

Taking the net with code 1: *Net-(C1-Pad1)* vs *+3V3*, and the report says: *ERROR:PCB net code 1 name mismatch (Net-(C1-Pad1) vs +3V3) (kibot.pre_update_xml - pre_update_xml.py:169)*

Is not the first time I see KiCad giving random numbers to things and then just ignoring the numbers. I'll try to adapt the code to this inconsistency.

set-soft, 2023-03-31:
Now this shouldn't be a problem.

jtmullen, 2023-03-31:
Thanks @set-soft, that is rather annoying on KiCad's part. It seems that even importing the netlist to the pcb (instead of updating from the schematic) doesn't change the numbering to match the netlist.

Fix is working, thanks!

commit 2f8e416f (2023-03-31): [PCB/SCH Parity] Added support for bogus net codes

Fixes #410