INTI-CMNB/KiBot #411: Installing debian package fails on current testing version 12 [bookworm]

ask6483, 2023-03-31:
```
Setting up kikit (1.3.0-5) ...
Traceback (most recent call last):
  File "/var/lib/dpkg/info/kikit.postinst", line 5, in <module>
    with open(fp, 'r') as f:
         ^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/kicad/template/fp-lib-table'
dpkg: error processing package kikit (--configure):
 installed kikit package post-installation script subprocess returned error exit status 1
```

set-soft, 2023-03-31:
Hi @ask6483 !

Do you have the *kicad-footprints* package installed?

ask6483, 2023-03-31:
No I'm using custom repository with footprints

ask6483, 2023-03-31:
Installed footprints, packages3d and templates but error is still there
```
kicad-demos/testing,testing,now 6.0.11+dfsg-1 all [installed]
kicad-doc-en/testing,testing,now 6.0.11+dfsg-1 all [installed]
kicad-footprints/testing,testing,now 6.0.11-1 all [installed]
kicad-packages3d/testing,testing,now 6.0.10-1 all [installed]
kicad-templates/testing,testing,now 6.0.8-1 all [installed]
kicad/testing,now 6.0.11+dfsg-1 amd64 [installed]
```

set-soft, 2023-03-31:
Are you sure that your *kicad-footprints* package doesn't contain */usr/share/kicad/template/fp-lib-table*. In stable, using backported KiCad, I get:

```
$ dpkg -S /usr/share/kicad/template/fp-lib-table
kicad-footprints: /usr/share/kicad/template/fp-lib-table
```

With the equivalent KiCad package:

```
$ dpkg -l kicad
Deseado=desconocido(U)/Instalar/eliminaR/Purgar/retener(H)
| Estado=No/Inst/ficheros-Conf/desempaqUetado/medio-conF/medio-inst(H)/espera-disparo(W)/pendienTe-disparo
|/ Err?=(ninguno)/requiere-Reinst (Estado,Err: mayúsc.=malo)
||/ Nombre         Versión               Arquitectura Descripción
+++-==============-=====================-============-============================================
ii  kicad          6.0.11+dfsg-1~bpo11+1 amd64        Electronic schematic and PCB design software
```

If the file is really missing then is some packaging error.  According to Debian it should be there: https://packages.debian.org/bookworm/all/kicad-footprints/filelist

ask6483, 2023-03-31:
Sorry next error was about missing '/usr/share/kicad/template/sym-lib-table'.
Installed kicad-symbols and kikat installation proceed now.

Last question: if I'm using non-default directories for footprints, packages3 and symbols it will have some negative impact on kibot functionality?

set-soft, 2023-03-31:
> Sorry next error was about missing '/usr/share/kicad/template/sym-lib-table'. Installed kicad-symbols and kikat installation proceed now.

Ok, I'm adding both as dependencies, so they are forced to be installed. This is needed because KiKit contains some symbols and footprints that are handy when using the KiKit multi-board mechanism. They could be a separated package, but I don't think this is worth the effort.  Lamentably KiCad doesn't have a mechanism to add libs in a clean way, KiCad was always a tool designed with Windows in mind (single user, mess of dependencies, etc.).

> Last question: if I'm using non-default directories for footprints, packages3 and symbols it will have some negative impact on kibot functionality?

You'll just need to be sure that KiBot is getting the paths right. Running it with debug information (i.e. -vvvv) will show which paths are used to look for libs. If you find something that can't be solved defining environment variables just report it. Also note that KiBot supports defining environment variables in the configuration file, so it gets self-contained (easier to deploy on other machine).

closed 2023-03-31 by set-soft

ask6483, 2023-03-31:
Thanks for your help.