INTI-CMNB/KiBot #412: [BUG] out.pcb_draw - Unsupported path element

arthursaunier, 2023-04-04:
Hi @set-soft 

I'm using the latest version of kibot (1.6.1) on WSL, and I get an error on my projects. 
I tried using kicad 6 and kicad 7, and I get the same error every time on PCBdraw.

My projects worked on the previous version of kibot (1.6.0), but now stopped working. 

```
ERROR:PcbDraw error: Unsupported path element 131.0000 (kibot - out_pcbdraw.py:500)
```

Here is a project where I have this problem, and the logs. 

[KiCad.zip](https://github.com/INTI-CMNB/KiBot/files/11146187/KiCad.zip)
[log.log](https://github.com/INTI-CMNB/KiBot/files/11146183/log.log)

Thanks for all your work!

set-soft, 2023-04-04:
Hi @arthursaunier !

Use the `dev` code. This issue is already solved there. This problem was reported to upstream 2 weeks ago: https://github.com/yaqwsx/PcbDraw/issues/142

You can also workaround it as I explain in the issue: replace the edge cut but lines instead of a plygon.

Is not a problem with 1.6.1. Is just that you used a rectangle (any polygon will be a problem) and KiCad 7.0.1 (which  generates polygons as polygons (not lines) in the SVG)

closed 2023-04-04 by set-soft

arthursaunier, 2023-04-04:
Thanks for your timely response and sorry about that, I didn't see the previous issue that solved it!