INTI-CMNB/KiBot #414: [FEATURE] download_model with header

N0cynym, 2023-04-04:
**Is your feature request related to a problem? Please describe.**

If I want to accsess a not public URL I need to add a PRIVATE-TOKEN to my request header.

**Describe the solution you'd like**

Add to the [download_model](https://github.com/INTI-CMNB/KiBot/blob/6ace82e20e1800bee53e98967fdbe897d3e1aef8/kibot/out_base_3d.py#L100) optional a privat token to be able to accsess private URLs.

set-soft, 2023-04-04:
Hi @N0cynym !
I need some example to test it. Not your token, but more information. Which service are you using? Adding `PRIVATE-TOKEN` looks too specific to one service, we should add some *header customization*.

N0cynym, 2023-04-04:
I have the case that I want to access to my GitLab repository which is privat.

If I create a token I could access via the API to my repository and do the same as `kicad_3d_url` is doing currently.

[image]

set-soft, 2023-04-05:
Hi @N0cynym !

I'm not sure if I understand. Are you currently passing the token using the URL? 
Something like:

`http://www.domain.com/index.html?PRIVATE-TOKEN=MYTOKEN`

If this is the case, why do you need another mechanism? If you are concerned about including your toke in the KiBot configuration, which is a really bad idea, read the *Doing YAML substitution or preprocessing* in the documentation.

As you can change the values using command line arguments you can hide your token in an environment variable, which is the common practice. 

If this works I don't see why we should add more complexity.

N0cynym, 2023-04-06:
Dear @set-soft I think the way how you discripe it works.

`http://www.domain.com/index.html?PRIVATE-TOKEN=MYTOKEN`

I thought that I can not access to my privat URL and was not sure how the download module  of python works.

So my idea was that the download module is missing the `PRIVATE-TOKEN` and it needs to be added seperatly as argument like:

`curl -H "PRIVATE-TOKEN: MYTOKEN" -o /root/archive.tar.gz https://gitlab.local.com/api/v4/projects/123/repository/archive?ref=master`

But I think my problem is that I accsess to the gitlab api incorrectly...

`https://gitlab.local.com/api/v4/projects/123/repository/archive/mod/packages3d/?ref=master?PRIVATE-TOKEN=MYTOKEN'`

set-soft, 2023-04-06:
I think the variables separator is `&`

https://gitlab.local.com/api/v4/projects/123/repository/files/mod/packages3d/?ref=master&PRIVATE-TOKEN=MYTOKEN

The `?` starts the variables, the variables are separated by `&`

N0cynym, 2023-04-12:
Dear @set-soft thank you for helping me with how to work with [query string](https://en.wikipedia.org/wiki/Query_string).

But unfortunately kibot is not building a gitlab api URL for me.

If I add: 

```yaml
options:
      pcb3d:
        kicad_3d_url: 'https://git.local.com/api/v4/projects/123/repository/files/mod%2Fpackages3d%2F?ref=master&private_token=MYTOKEN'
```

KiBot builds:

```
WARNING:(W048) Failed to download `https://git.local.com/api/v4/projects/123/repository/files/mod%2Fpackages3d%2F?ref=master&private_token=MYTOKEN-[MASKED]/R_0805_2012Metric.wrl` (kibot - out_base_3d.py:163)
WARNING:(W047) Missing 3D model for R1: `${KISYS3DMOD}/Resistor_SMD.3dshapes/R_0805_2012Metric.wrl` (kibot - out_base_3d.py:498)
```

Correctly working does this:
```
curl "https://git.local.com/api/v4/projects/108/repository/files/mod%2Fpackages3d%2FResistor_SMD.3dshapes%2FR_0805_2012Metric.wrl?ref=master&private_token=MYTOKEN"
```

I noted that the API documentation requires 

<html><body>

Attribute | Type | Required | Description
-- | -- | -- | --
file_path | string | yes | URL encoded full path to new file, such as lib%2Fclass%2Erb.

</body>
</html>

Do you have any advices for me?

set-soft, 2023-04-13:
Hi @N0cynym !

I see the problem, you need to insert the file name, not concatenate it. I can add such an option. But the whole usage case doesn't look good to me.

Why do you need to ask GitLab for a file? Isn't this file part of some repo?  If this is the case: why don't you just include the second repo as a submodule? Git can handle repos that includes other repos, this is how most people handle KiCad libs shared by various repos, not manually downloading the files.

N0cynym, 2023-04-14:
@set-soft thanks for your time looking into my issue.

I'm having my KiCad Lib in a self hosted Gitlab repo to share it with people and use/share the same lib.

All projects are also in the self hosted Gitlab.

With the help of your project I'm able to use as less as necessary storage for each project and to export all very efficient.

I'm trying to access my libray project in gitlab to avoid to much memory traffic and reduce pipeline time.
So I think that if I use kibot and download just the needed 3d files is better as downloading my whole lib project as submodule each time I run a pipeline.
Do you dissagree with my look at it?
Can you help me to find the most efficent way?

Here is how I want to let it work:

[image]

set-soft, 2023-04-14:
You can divide your lib into units that aren't heavy.

Anyways, I added an option called `kicad_3d_url_suffix` you can use it to add the GET variables there. The resulting URL is:

URL name_of_the_model_quoted SUFFIX

N0cynym, 2023-04-18:
Dear @set-soft,
Sorry for testing very lately.

The URL is built correctly but the `private_token` is cutted.

Error URL
```
https://git.local.com/api/v4/projects/123/repository/files/mod%2Fpackages3d%2FCapacitor_SMD.3dshapes%2FC_0603_1608Metric.wrl?ref=master&private_token=[MASKED]abcd-
```

Config example
```yaml
        kicad_3d_url: 'https://git.local.com/api/v4/projects/123/repository/files/mod%2Fpackages3d%2F'
        kicad_3d_url_suffix: '?ref=master&private_token=abcde-12345678901234567890'
```

N0cynym, 2023-04-18:
@set-soft 

Note: For me it looks like that you build the URL correctly.

But [GitLab is masking](https://gitlab.com/gitlab-org/gitlab/-/issues/350041) the token with secrect detection.

I'm looking forward to find a solution with the current KiBot features.

If you have some advices it would be kind to share them with me. ☺️

set-soft, 2023-04-20:
Hi @N0cynym !

> The URL is built correctly but the `private_token` is cutted.

How do you verify it? I mean: from where did you copy `https://git.local.com/api/v4/projects/123/repository/files/mod%2Fpackages3d%2FCapacitor_SMD.3dshapes%2FC_0603_1608Metric.wrl?ref=master&private_token=abcd-[MASKED]`?

If this is from the GitLab web interface, looking at the logs for your workflow, this doesn't really mean the *[MASKED]* text was sent to any server. At least on GitHub all the logs you access from the web interface are censured. Take a look to [KiBot actions](https://github.com/INTI-CMNB/KiBot/actions/runs/4733699841/jobs/8401628982), expand the "Trigger garbage collector" task, you'll see `Run curl -X POST -u "***:***" -` these asterisks are the user and token used in the web request. They are hidden to protect the secrets. What is strange of your text is that *abc-* isn't masked.

You must look at logs in a file. I.e. run KiBot with debug and redirect it to a file, then include the file in the artifacts and download the artifacts.

N0cynym, 2023-04-21:
So I verified this in the GitLab Web interface after letting KiBot print debug logs.

`private_token=abcd-[MASKED]` is in real output `private_token=glpt-[MASKED]` and `glpt` seems to mean **G**it**L**ab **P**rivat **T**oken

I think I should try to let the pipeline write the KiBot debug into a file and save this as artifact.

Hopefully this will bring more light into my issue with GitLab secret detection.

Thanks for your advice and help. I'll test a bit more but this feature seems to definitely built the URL for downloading correctly.

set-soft, 2023-06-15:
Any news about this?

N0cynym, 2023-06-16:
@set-soft unfortunately I still seem to have a issue at gitlab.
I'll post into this issue if I found my solution. 

The feature I requested does work as it should. 👍 

It's just my gitlab pipeline which needs to be finished.

closed 2023-06-16 by N0cynym