INTI-CMNB/KiBot #416: [FEATURE] Add ability to render specific ViewPort in 3D viewer

Kedarius, 2023-04-06:
Would it be possible to extend the render_3d to use specific view port from 3D viewer? I have a KiBot integrated into CI (TeamCity) and one of the problem is that one needs to specify the 3D viewer settings in the config and I like to have one config for all projects. So the PCBs are small or not whole or incorrectly rotated....
And so it would be great to have an ability to save the viewport in KiCad and have some well known names in KiBot config (i.e. front, back, product-overview,....).

set-soft, 2023-04-06:
I don't think it will be possible using KiCad. 

In fact some KiCad developers told me to forget about a reliable 3D View interface in the near future. Take a look at the following [feature request](https://gitlab.com/kicad/code/kicad/-/issues/13948) 

It means that using KiCad we don't have a good control of the view.

But I think you should try using the `blender_export` output. You'll much more predictable results, by default it adds a camera that points to board and makes it big in the image. Is much more easy to get a good render using this output.

I know Blender is really big, but is part of the "full" docker images. And the quality difference well deserves the extra time it takes to render. Note that Jon Evans, one of the most prolific KiCad developers, suggested to always use Blender.

The light is the only tricky part. The one added by Kibot works well in most cases. But I found a board where it doesn't. But looks like something workable. 

You can get top and bottom views just specifying them, no need for zoom or x,y movement. And using a simple rotation (X=30º Z=-20º) you get: [image]

This rotation makes the board to face the light, intentionally positioned at the left.

I think this is what you are looking for.

Kedarius, 2023-04-06:
Thanks for the reply. I will look into the blender export a little bit more :-) 
Right now I am getting errors about some mismatch between KICAD6_3DMODEL_DIR a KICAD7_3DMODEL_DIR, probably because I have like 5 KiCADs installed on my workstation....

closed 2023-04-06 by Kedarius

set-soft, 2023-04-06:
The KiCad 3D models in v7 is a mess the Linux release for 7.0.0/1 has all the footprints pointing to KiCad 6 3D models.

The 7.0.0 DRC Python API doesn't load the global fp-lib-table, fixed in 7.0.1.

So you have to be patient and play with the environment variables a little bit.

Kedarius, 2023-04-06:
I made a separate issue for that ( #417 ) because I am not able to get it working even in the docker images....