INTI-CMNB/KiBot #417: [HOW] I am trying to get step and blender export working

Kedarius, 2023-04-06:
**What do you want to achieve?**
I am trying to get 3D exports (STEP and Blender) working. I have a simple project with some logic and some LED bars and 7-segment displays. In KiCad 7.0.1 (on my Windows PC) it looks like this:
[image]

**Do you have some PCB/Schematic to use as example?**
The project in question [multidisp.zip](https://github.com/INTI-CMNB/KiBot/files/11173174/multidisp.zip)

**Do you have some configuration file (.kibot.yaml) that you are using?**
[kibot-development.yaml.zip](https://github.com/INTI-CMNB/KiBot/files/11173180/kibot-development.yaml.zip)

**Environment (please complete the following information):**
Where are you running KiBot:
I am running KiBot on a Linux server (Fedora 36) in Docker via TeamCity (pulling from a local private repo). I've used ghcr.io/inti-cmnb/kicad7_auto_full:latest a ghcr.io/inti-cmnb/kicad7_auto_full:dev image with no visible differences.

I tried to setup the 3D cache folder and run STEP export before Blender export and it did download 3 files - LED_0603_1608Metric.wrl, C_0805_2012Metric.wrl and R_0603_1608Metric.wrl. But not even these files appear in final STEP export.

The step export finishes and produces file (but with errors and the step file is incomplete) and the blender does not produce any output.
Here is the log:
<pre>
- Updating BoM in XML format
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/7.0/kicad_common.json (eeschema_do - misc.py:198) (kibot - kiplot.py:134)
- 'Exports the PCB as a 3D model.' (step) [step]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:342)
WARNING:(W010) Unable to find KiCad user templates (kibot - config.py:417)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:417)
ERROR:Unable to expand `KICAD7_3DMODEL_DIR` in `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - config.py:108)
WARNING:(W011) Missing default symbol library table (kibot - config.py:493)
ERROR:/usr/share/kicad/footprints/*.pretty (kibot - config.py:496)
WARNING:(W047) Missing 3D model for U2: `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J4: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x08_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for U4: `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J6: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x02_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for U1: `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J3: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x08_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J2: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x08_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J1: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x14_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for U3: `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for J5: `${KICAD7_3DMODEL_DIR}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x08_P2.54mm_Vertical.wrl` (kibot - out_base_3d.py:490)
WARNING:(W047) Missing 3D model for SW1: `${KICAD7_3DMODEL_DIR}/Button_Switch_THT.3dshapes/SW_DIP_SPSTx04_Slide_9.78x12.34mm_W7.62mm_P2.54mm.wrl` (kibot - out_base_3d.py:490)
WARNING:(W098)  3 3D models downloaded (kibot - out_base_3d.py:503)
- 'Exports the PCB in blender format' (blender_3d_export) [blender_export]
ERROR:Unknown KiCad dialog: KiCad PCB Editor Error (pcbnew_do - interposer.py:405) (kibot - kiplot.py:131)
ERROR:Potential dialog messages: {"Failed to retrieve file times for '' (error 2: No such file or directory)", '17:29:37', 'OK', 'H', 'Details', '17:29:36'} (pcbnew_do - interposer.py:406) (kibot - kiplot.py:131)
ERROR:/usr/bin/pcbnew_do returned 9 (kibot - kiplot.py:201)
</pre>
And here is the produced STEP file:
[step.zip](https://github.com/INTI-CMNB/KiBot/files/11173257/step.zip)
And it looks like this:
[image].

I tried to manually copy all the mentioned files (and STEP substitutes for wrmls) into the 3D cache folder but the result is the same.

I have no idea what I am doing wrong or how should I setup the 3D exports to work properly.

set-soft, 2023-04-06:
The code to download 3D models didn't pay attention to KiCad 7 models. Now they should get downloaded.

Copying them to the 3D cache didn't work because they are picked by the downloader.

closed 2023-04-06 by set-soft

Kedarius, 2023-04-06:
Should it be already working in the dev docker images? Because I pulled the dev image:
<pre>
# docker pull ghcr.io/inti-cmnb/kicad7_auto_full:dev
dev: Pulling from inti-cmnb/kicad7_auto_full
......
Digest: sha256:102b5381b4d15dbd9abc830baaf36017e7fa10b04dec4ad7a38a2384e7633757
Status: Downloaded newer image for ghcr.io/inti-cmnb/kicad7_auto_full:dev
</pre>
And that fixed the STEP export but the blender one still complains:
<pre>
- 'Exports the PCB in blender format' (blender_3d_export) [blender_export]
ERROR:Unable to expand `KICAD7_3DMODEL_DIR` in `${KICAD7_3DMODEL_DIR}/Display_7Segment.3dshapes/HDSP-7401.wrl` (kibot - config.py:108)
WARNING:(W098)  8 3D models downloaded or cached (kibot - out_base_3d.py:507)
ERROR:Unknown KiCad dialog: KiCad PCB Editor Error (pcbnew_do - interposer.py:416) (kibot - kiplot.py:131)
ERROR:Potential dialog messages: {'22:48:32', 'Details', "Failed to retrieve file times for '' (error 2: No such file or directory)", 'OK', 'H'} (pcbnew_do - interposer.py:417) (kibot - kiplot.py:131)
ERROR:/usr/bin/pcbnew_do returned 9 (kibot - kiplot.py:201)
</pre>

set-soft, 2023-04-07:
The code you used is the one fixed because it says "3D models downloaded or cached", the old code just said "downloaded", but the cached models are in the same list.

I already told you to enable debug, without debug is impossible to know what's exactly going on.

The problems is KiCad creating a dialog with error, perhaps some problem with the models. But can't be sure without more information.  

Trying to reproduce it on a docker image I found it looks like a KiCad bug. Try the following with your design:

1. Open the PCB Editor
2. Choose the File | Export | VRML option
3. Make sure the "Copy 3D model files ..." and "Use relative paths ..." and enabled
4. Press OK

You'll get a dialog named "KiCad Error" with the message reported by KiAuto "Failed to retrieve file times for '' ..."

Here we hit two bugs in KiCad:

1. Paths relative to the footprint can't be used for VRML export. The relative paths are a very nice feature, but they are quite in beta, I wonder why they included it in the release knowing it.
2. KiCad can't even say the name of the file that failed to find.

I'll try to workaround it, KiBot can rename models, so we can change the name to avoid the relative issue.

Kedarius, 2023-04-07:
> I already told you to enable debug, without debug is impossible to know what's exactly going on.
> 

I am not sure, where you did say that but I probably did miss it somehow. I provided the whole project and all configs. But here is the log of the failing run: [make_outputs-debug-2231569.zip](https://github.com/INTI-CMNB/KiBot/files/11179508/make_outputs-debug-2231569.zip) . However the issue seems to be resolved.

> The problems is KiCad creating a dialog with error, perhaps some problem with the models. But can't be sure without more information.
> 
> Trying to reproduce it on a docker image I found it looks like a KiCad bug. Try the following with your design:
> .....
> You'll get a dialog named "KiCad Error" with the message reported by KiAuto "Failed to retrieve file times for '' ..."

Yeah. With this I was able to find which components cause the problem. And the issue is that somehow KiCad decided to change the path to relative i.e. ../74HC238D,653.stp . I tried to change it to ${KIPRJMOD}/logic/74HC238D,653.stp and the error in KiCad went away.

> Here we hit two bugs in KiCad:
> 
>     1. Paths relative to the footprint can't be used for VRML export. The relative paths are a very nice feature, but they are quite in beta, I wonder why they included it in the release knowing it.
And the thing is that KiCad changed it somehow, because the original footprint (downloaded from Mouser) has the step file specified without path at all. 

However the issue seems to be resolved - it created a file that I have been able to open in Blender. Now I need to play with the config and maybe with Blender, because the PNG is black as night (probably I am missing a light configuration, ....).

Thanks for your help.

set-soft, 2023-04-07:
Ok, the above patch converts any path relative to the footprint in an absolute path.

So now KiCad can find the models.

I got:

[image]

Using:

```yaml
  - name: 'blender_3d_export'
    comment: "Exports the PCB in blender format"
    type: blender_export
    dir: '3D/blender'
    options:
      render_options:
        transparent_background: true
        samples: 10
        resolution_x: 1920
        resolution_y: 1080
      point_of_view:
        rotate_x: -30
        rotate_z: 160
      outputs:
        - type: blender
        - type: render
```

1. Don't try to manually set a camera from the beginning, this is something you do loading the blender file.
2. I used -30º 160º so the text in your board is easier to read. (30 -> -30, -20+180 -> 160)

set-soft, 2023-04-07:
> I tried to change it to ${KIPRJMOD}/logic/74HC238D,653.stp and the error in KiCad went away.

This is the simplest solution. But it prevents from moving the footprint to other folder. The idea of path models relative to the footprints is a good idea (I wonder why it wasn't this way from the beginning) but the current implementation needs more debugging.

Kedarius, 2023-04-07:
> This is the simplest solution. But it prevents from moving the footprint to other folder. The idea of path models relative to the footprints is a good idea (I wonder why it wasn't this way from the [...]

I agree. And the second issue is the KiCad inconsistency as one export (STEP) does it in a one way and VRML does it differently. 
The relative path would be nice as I am trying to have all the libraries as git submodules and with the KIPRJMOD way they need to be in the same path in the project. However I can live with that. Electronics is more a hobby and side job for me, so I do not need to have it perfectly perfect :-) 

> * Don't try to manually set a camera from the beginning, this is something you do loading the blender file.
> 
> * I used -30º 160º so the text in your board is easier to read. (30 -> -30, -20+180 -> 160)
Yeah I will start from your config. I did get mine somewhere and it got all complicated as I was trying to figure out the problem.

Another issue is that the renders on my CI server are taking ages as it does not have a GPU, so it looks like I will look into Blender a little bit more :-)

Once more thanks for your help, I've been able to render the same image as you....

set-soft, 2023-04-07:
> The relative path would be nice as I am trying to have all the libraries as git submodules and with the KIPRJMOD way they need to be in the same path in the project. However I can live with that. 

With the [52ac92c](https://github.com/INTI-CMNB/KiBot/commit/52ac92c42dbf55f4bb159e9ab33c010473961d01) patch you can safely use the relative paths. KiBot will convert them to absolute in the temporal file passed to KiCad.

 
> Another issue is that the renders on my CI server are taking ages as it does not have a GPU, so it looks like I will look into Blender a little bit more :-)

You don't need a GPU to get a decent render in about 1 minute. Of course, with an RTX board Blender can do it in real time. It just needs enough memory. My 2nd generation Ryzen 5 needs 1m 10s to export the board and create the above render. This CPU was released in april 2018. I never tried using my GPU (a Radeon board) because I never installed the AMD drivers, just use the free software drivers.

Kedarius, 2023-04-07:
Well my CI server is a VM with 2 cores (some 2-3 years old Xeon) and 4GB RAM.... And it takes like 5 minutes to render the image. I can increase the resources if needed. But also I can look into Blender a little deeper, every opportunity to learn something new is a good reason to do so :-)

set-soft, 2023-04-10:
> 4GB RAM....

This could be the problem, try with more memory.

Kedarius, 2023-04-13:
I know it is a little bit off topic for this issue but on the other hand it can be useful for someone who finds this issue in search for a blender renders related info.

You can use a -vvv option to get blender output and from that I can see that this relatively simple board takes up to 700MB of RAM max. 
When rendering directly in Blender on my local Windows desktop with Intel's 12th gen CPU and 2060RTX GPU, it takes about a few seconds using GPU and about 10 seconds using CPU. However on my older laptop with 1080 GPU and 7th gen i5 it takes about 4 minutes on CPU and about 15 seconds on GPU. 

So basically I've concluded that using GPU is preferable however modern CPUs can be comparable fast while older CPUs are not.

commit 6649038d (2023-04-06): [3D][Fixed][KiCad 7] 3D models not downloaded

Closes #417