INTI-CMNB/KiBot #418: [BUG] Unknown kicad_sch attribute `[Symbol('netclass_flag')

N0cynym, 2023-04-14:
**Describe the bug**
I'm trying to use the new feature [netclass](https://docs.kicad.org/master/en/eeschema/eeschema.html#schematic-netclasses).
KiBot seems not to know about them.

**To Reproduce**
Add Netclass.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow `GitLab`
- Using docker image ghcr.io/inti-cmnb/kicad7_auto:dev 

**Additional context**

```
ERROR:Unknown kicad_sch attribute `[Symbol('netclass_flag'), '', [Symbol('length'), 2.54], [Symbol('shape'), Symbol('rectangle')], [Symbol('at'), 222.25, 41.91, 0], [Symbol('effects'), [Symbol('font'), [Symbol('size'), 1.27, 1.27]], [Symbol('justify'), Symbol('left'), Symbol('bottom')]], [Symbol('uuid'), Symbol('356c46fb-1c9f-46ea-82f7-5ade23c8fdc8')], [Symbol('property'), 'Netclass', 'NetCl_power', [Symbol('at'), 222.25, 38.1, 0], [Symbol('effects'), [Symbol('font'), [Symbol('size'), 0.635, 0.635], Symbol('italic')]]]]` (kibot - kiplot.py:270)
```

set-soft, 2023-04-14:
Hi @N0cynym !

KiCad schematic format documentation doesn't mention anything about netclasses.

Moreover, netclasses are stored in the project.

So I think this is some additional feature, I need an example project using it.

N0cynym, 2023-04-14:
This should fail if you anything with kibot.

[Test-418.zip](https://github.com/INTI-CMNB/KiBot/files/11234626/Test-418.zip)

set-soft, 2023-04-14:
This should work now. They are very similar to other labels, just adds a length.

I didn't see these flags. Note that they aren't mandatory, you can tag net without using these flags.

closed 2023-04-14 by set-soft

N0cynym, 2023-04-14:
Works, thanks you!

commit 3b0a26c7 (2023-04-14): [Added] Support for netclass flags

Closes #418