INTI-CMNB/KiBot #419: [BUG] VRML export options ref_x and ref_y doesn't work as expected

rainer38, 2023-04-15:
**Describe the bug**
The options "ref_x" and "ref_y" of output type "vrml" doesn't work as expected.
The option "ref_y" has no effect at all.
The option "ref_x" always shift in x and y direction the same amount of units.

I compared the VRML files and it shows the coordinate offset for ref_y is ignored and for ref_x is applied to x and y coordinates

**To Reproduce**
I use the following yaml configuration:
```
kibot:
  version: 1

global:
  dir: /mnt/ramdisk/test/
  units: millimeters

outputs:
- name: export_vrml
  comment: PCB to VRML export
  type: vrml
  options:
    output: '%f_v%br_%I.%x'
    download: false
    dir_models: ''
    use_pcb_center_as_ref: false
    ref_x: 110.0
    ref_y: 90.0
```

The problem is reproducible with every KiCAD board, no special requirements.
After i import the VRML file in Blender or FreeCAD the placement of the objects are not correct.

**Expected behavior**
When i use ref_x=110 and ref_y=90 i expect the board to be shifted 110 units in x direction and 90 units in y direction within the VRML file.

**Screenshots**
In the KiCAD pcb the lower left corner of the Edge.Cut Layer is at 110,90.
When i set ref_x=0 and ref_y=0 the lower left corner of the imported pcb is at the expected position (110,-90)
[image]

When i set ref_x=0 and ref_y=90 the lower left corner of the imported pcb is still at the position (110,-90) which is wrong.
[image]

When i set ref_x=110 and ref_y=0 the lower left corner of the imported pcb is at the position (0,20) which is wrong.
[image]

When i set ref_x=110 and ref_y=90 the lower left corner of the imported pcb is at the position (0,20) which is wrong.
[image]

The difference between the options refx=0, ref_y=0 and ref_x=110, ref_y=0 is an offset of exact 110 in x and y direction.
[image]

**Environment (please complete the following information):**
I use the deb packages on Debian bullseye and manually run kibot on the command-line.

KiBot 1.6.1 - Copyright 2018-2023, Salvador E. Tropea/INTI/John Beard - License: AGPL v3

Application: KiCad
Version: (7.99.0-731-g32eb1ef77e-dirty), release build
Libraries:
	wxWidgets 3.0.5
	FreeType 2.10.4
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.88.1 GnuTLS/3.7.1 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3
Platform: Debian GNU/Linux 11 (bullseye), 64 bit, Little endian, wxGTK, KDE, x11
Build Info:
	Date: Apr 10 2023 18:30:30
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.88.1
	ngspice: 39
	Compiler: GCC 10.2.1 with C++ ABI 1014
Build settings:
	KICAD_SPICE=ON

set-soft, 2023-04-15:
Thanks for the detailed explanation.

The code was incomplete. Now should work.

closed 2023-04-15 by set-soft

rainer38, 2023-04-16:
Now the coordinates are where they should be. Nice work.

commit 5e80a1e1 (2023-04-15): [VRML][Fixed] Y coordinate

Fixes #419