INTI-CMNB/KiBot #420: [FEATURE] Add "use_aux_axis_as_origin" for VRML output

rainer38, 2023-04-15:
**Is your feature request related to a problem? Please describe.**
When i export the VRML file i manually have to set the ref_x and ref_y coordinates to my aux origin from the pcb to set the correct origin in the exported file.

**Describe the solution you'd like**
Add an option "use_aux_axis_as_origin" like with other output types to set the ref_x and ref_y automatically to the aux origin of the pcb.

**Describe alternatives you've considered**
I've written al little python script which parses the KiCAD pcb file to get the aux origin and modify the kibot.yaml with the result. But this would mean i have to generate a specific kibot.yaml for every project instead of using one global config file.

set-soft, 2023-04-15:
Hi @rainer38 !
Try the above patch

closed 2023-04-15 by set-soft

rainer38, 2023-04-16:
This also works very good.

commit 3c749719 (2023-04-15): [VRML][Added] Option to use the auxiliary origin as reference

Closes #420