INTI-CMNB/KiBot #421: [FEATURE] Please add a variable for the main folder name of the KiCAD project

rainer38, 2023-04-15:
**Is your feature request related to a problem? Please describe.**
I need a variable which contains the main KiCAD project folder name without the whole path.
For example if the project path would be "/home/user/Documents/kicad_project/testfile.kicad_pcb" the variable would contain "kicad_project".
My projects does not always have the same file- and folder name but the folder name is the one i need for generating output files.

**Describe the solution you'd like**
Please add a variable like %M (or something else) which represents the folder name of the KiCAD project.

**Describe alternatives you've considered**
I wrote a little python script which extracts the folder name and substitutes placeholders within the kibot.yaml file. But than i have to create custom configuration files for every project instead of using a global configuration.

set-soft, 2023-04-15:
Hi @rainer38 !
The above patch should do it, please test it. It passed a naive test.

closed 2023-04-15 by set-soft

rainer38, 2023-04-16:
It works like a charm. Thanks for the fast reaction. :)

commit 124fc8e7 (2023-04-15): [Added] Expansion pattern **%M**

- Directory where the pcb/sch resides. Only the last component
  i.e. /a/b/c/name.kicad_pcb -> c

Closes #421