INTI-CMNB/KiBot #422: [PROBLEM] Exception when running some output types

rainer38, 2023-04-15:
**What do you want to achieve?**
I try to generate excellon drill files and use pcb_print for generating pdf files but always get an exception thrown. Other output types work as expected. I don't know if the problem comes from my KiCAD installation or some other incompatibilities. I don't think this is a bug in kibot but maybe a changing break in newer KiCAD code.

**Do you have some configuration file (.kibot.yaml) that you are using?**
```yaml
kibot:
  version: 1

global:
  dir: /mnt/ramdisk/test/
  units: millimeters

outputs:
- name: gerber_drill
  comment: Drillfile
  type: excellon
  options:
    use_aux_axis_as_origin: true

- name: pcb_pdf
  comment: PCB as PDF
  type: pcb_print
  options:
    format: PDF
    pages:
    - layers:
      - layer: F.Cu
      - layer: F.Mask
        color: '#14332440'
      - layer: F.Paste
      - layer: F.Silkscreen
      - layer: Edge.Cuts
      sheet: Front copper
```

**Problem description**
When i run "kibot gerber_drill" i get the following exception:
```
Using config file: export.kibot.yaml
Using SCH file: testprojekt.kicad_sch
- 'Drillfile' (gerber_drill) [excellon]
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.1', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 407, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 521, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 512, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 423, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 192, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_any_drill.py", line 180, in run
    files = self.get_file_names(output_dir)
  File "/usr/lib/python3/dist-packages/kibot/out_any_drill.py", line 140, in get_file_names
    self._configure_writer(GS.board, wxPoint(0, 0))
  File "/usr/lib/python3/dist-packages/kibot/out_excellon.py", line 41, in _configure_writer
    drill_writer.SetOptions(self.mirror_y_axis, self.minimal_header, GS.p2v_k7(offset), self.pth_and_npth_single_file)
  File "/usr/lib/python3/dist-packages/kibot/gs.py", line 274, in p2v_k7
    return pcbnew.VECTOR2I(point) if GS.ki7 else point
  File "/usr/local/lib/python3.9/dist-packages/pcbnew.py", line 4995, in __init__
    _pcbnew.VECTOR2I_swiginit(self, _pcbnew.new_VECTOR2I(*args))
TypeError: in method 'new_VECTOR2I', argument 1 of type 'VECTOR2< int > const &'
Additional information:
Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.
  Possible C/C++ prototypes are:
    VECTOR2< int >::VECTOR2()
    VECTOR2< int >::VECTOR2(int,int)
    VECTOR2< int >::VECTOR2(VECTOR2< int > const &)
```

And when i run "kibot pcb_pdf" i get:
```
Using config file: export.kibot.yaml
Using SCH file: testprojekt.kicad_sch
- 'PCB as PDF' (pcb_pdf) [pcb_print]
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.1', 'console_scripts', 'kibot')())
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 407, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 521, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 512, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 423, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 192, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_pcb_print.py", line 1173, in run
    self.generate_output(output)
  File "/usr/lib/python3/dist-packages/kibot/out_pcb_print.py", line 1123, in generate_output
    self.plot_frame_internal(pc, po, p, len(pages)+1, len(self.pages))
  File "/usr/lib/python3/dist-packages/kibot/out_pcb_print.py", line 513, in plot_frame_internal
    ws.draw(GS.board, self.cleared_layer, page, self.paper_w, self.paper_h, tb_vars)
  File "/usr/lib/python3/dist-packages/kibot/kicad/worksheet.py", line 543, in draw
    e.draw(self)
  File "/usr/lib/python3/dist-packages/kibot/kicad/worksheet.py", line 191, in draw
    e.draw_line(p, st, en)
  File "/usr/lib/python3/dist-packages/kibot/kicad/worksheet.py", line 173, in draw_line
    s.SetStart(GS.p2v_k7(st))
  File "/usr/lib/python3/dist-packages/kibot/gs.py", line 274, in p2v_k7
    return pcbnew.VECTOR2I(point) if GS.ki7 else point
  File "/usr/local/lib/python3.9/dist-packages/pcbnew.py", line 4995, in __init__
    _pcbnew.VECTOR2I_swiginit(self, _pcbnew.new_VECTOR2I(*args))
TypeError: in method 'new_VECTOR2I', argument 1 of type 'VECTOR2< int > const &'
Additional information:
Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.
  Possible C/C++ prototypes are:
    VECTOR2< int >::VECTOR2()
    VECTOR2< int >::VECTOR2(int,int)
    VECTOR2< int >::VECTOR2(VECTOR2< int > const &)
```

In both cases the error comes from \_\_init\_\_ in /usr/local/lib/python3.9/dist-packages/pcbnew.py which comes from my self-compiled KiCAD version. It s compiled at April 10 and the newest code from gitlab. Maybe it broke something which works in the official released 7.0.1 version but i don't know how to debug this problem.

**Environment (please complete the following information):**
KiBot 1.6.1 - Copyright 2018-2023, Salvador E. Tropea/INTI/John Beard - License: AGPL v3

Application: KiCad
Version: (7.99.0-731-g32eb1ef77e-dirty), release build
Libraries:
	wxWidgets 3.0.5
	FreeType 2.10.4
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.88.1 GnuTLS/3.7.1 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3
Platform: Debian GNU/Linux 11 (bullseye), 64 bit, Little endian, wxGTK, KDE, x11
Build Info:
	Date: Apr 10 2023 18:30:30
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.88.1
	ngspice: 39
	Compiler: GCC 10.2.1 with C++ ABI 1014
Build settings:
	KICAD_SPICE=ON

set-soft, 2023-04-15:
Hi @rainer38 !

Try with the current development code. The 7.99 changes all the time and some incompatibilities are newer than KiBot 1.6.1.  

Even using the development code of KiBot you'll face problems with 7.99 periodically. The KiCad developers don't warranty compatibility for the Python API and the new kicad-cli tool. In fact they break things even for the stable release. KiCad 7.0.2 has at least 3 bugs that affect KiBot. I found them just while they were releasing it. And the above mentioned API and tool are low priority for KiCad developers.

That said, I'll try to keep 7.99 working, just report incompatibilities. I'm currently creating a docker image with the KiCad nightly so I can test fresh 7.99 code.

set-soft, 2023-04-15:
Looking at the logs this problem is already "fixed" by 1ea51f490891d5ba3939148afc4eb47e4de40813

KiCad 7.99 eliminated the constructor that takes a wxPoint (found in 7.0.x). So now we pass the x and y separated.

You need 1.6.2 from git

closed 2023-04-15 by set-soft

rainer38, 2023-04-16:
I checked out the dev branch from the git version and compiled it as a deb package.
After i installed the new 1.6.2 over my 1.6.1 now everything works as expected. :)

But i found a few files in the folder "tests/reference/8_0_0/" of the dev branch which have absolute symlinks instead of relative. I had to fix them to successfully compile the package.

set-soft, 2023-04-16:
> But i found a few files in the folder "tests/reference/8_0_0/" of the dev branch which have absolute symlinks instead of relative. I had to fix them to successfully compile the package.

Thanks, didn't notice they were wrong. They are new and very experimental.