INTI-CMNB/KiBot #424: [BUG]

Informaticore, 2023-04-16:
Running the following KiBot workflow crashes with an error. I got the workflow together with the KiCad CI Template from nerdyscout. I changed the docker image to run the latest kicad7 image.

Workflow sample:
```
  render:
    runs-on: ubuntu-latest
    needs: [DRC]
    container:
      image: ghcr.io/inti-cmnb/kicad7_auto:latest
    steps:
      - uses: actions/checkout@v3
      - name: run kibot
        run: | 
          kibot -c kibot.yml -e ${{env.schema}} -b ${{env.board}} -d ${{env.dir}} -s run_erc,run_drc -v \
            board_black_top.svg board_black_bottom.svg
      - uses: actions/upload-artifact@v3.1.1
        if: ${{success()}}
        with:
          name: ${{github.event.repository.name}}_img
          if-no-files-found: error
          path: |
            ${{env.dir}}/img/**
            !${{env.dir}}/**/*.ogv
            !${{env.dir}}/**/*.log
```

Error:
```
ERROR:PcbDraw error: Cannot locate resource /usr/share/pcbdraw/styles/set-black-cu.json; explored paths:
- /usr/share/kibot/pcbdraw
- /github/home/.local/share/pcbdraw
- /usr/share/pcbdraw (kibot - out_pcbdraw.py:500)
Error: Process completed with exit code 20.
```

somehow it can not find the set-black-cu.json.

set-soft, 2023-04-16:
The *.kibot/plot.kibot.yml* file is outdated, and currently wrong. I tries to workaround a veeeryyy old issue. It specifies an absolute path for the *style*. This isn't needed and in fact it currently fails because now *pcbdraw* is part of KiBot. You should report it to @nerdyscout 

The following change solves your proble:

```diff
diff --git a/.kibot/plot.kibot.yml b/.kibot/plot.kibot.yml
index ab3d22b..ec0939c 100644
--- a/.kibot/plot.kibot.yml
+++ b/.kibot/plot.kibot.yml
@@ -31,7 +31,7 @@ outputs:
       # The default is none
       show_components: none
       # [string|dict] PCB style (colors). An internal name, the name of a JSON file or the style options
-      style: '/usr/share/pcbdraw/styles/set-black-cu.json'
+      style: set-black-cu
       # [string=''] Board variant to apply
       variant: ''
       # [boolean=false] render V-CUTS on the Cmts.User layer
@@ -71,7 +71,7 @@ outputs:
       # The default is none
       show_components: none
       # [string|dict] PCB style (colors). An internal name, the name of a JSON file or the style options
-      style: '/usr/share/pcbdraw/styles/set-black-cu.json'
+      style: set-black-cu
       # [string=''] Board variant to apply
       variant: ''
       # [boolean=false] render V-CUTS on the Cmts.User layer
```

BTW: Now KiBot supports some sort of templates that can easily implement the @nerdyscout templates. So perhaps we could incorporate them.

closed 2023-04-16 by set-soft

Informaticore, 2023-04-16:
Sorry - I don't get it. Where do I have to change that? There is no plot.kibot.yml file. How is this supposed to be used anyway? I have the feeling that  the nerdyscout template is outdated anyway even the way how he is using the kibot.
We should maybe simply create a workflow github snipped which includes the parts which are really needed for generating manufacturing files like the thing from nerdyscout does.

set-soft, 2023-04-16:
> Sorry - I don't get it. Where do I have to change that? There is no plot.kibot.yml file. 

The diff says where is the file, please take a look at it. The link to your repo is https://github.com/open-thngs/dps368-breakout/blob/main/.kibot/plot.kibot.yml

> How is this supposed to be used anyway?

What? The pcbdraw output? Read the KiBot docs. The templates? Not sure, I know nerdyscout worked on it and did its best, but I'm not sure what he implemented, I guess its docs explains it.

> I have the feeling that the nerdyscout template is outdated anyway even the way how he is using the kibot.

Most probably, I try to keep things as compatible as possible, but sometimes some changes impact old configurations.

> We should maybe simply create a workflow github snipped which includes the parts which are really needed for generating manufacturing files like the thing from nerdyscout does.

Please read the KiBot README, it will explain various approaches. The most simple is using the quick start and extract the parts you want from the example. Most people does it. You'll find repo examples using it manually and using GitHub actions.

nerdyscout, 2023-04-17:
> > I have the feeling that the nerdyscout template is outdated anyway even the way how he is using the kibot.
>
> Most probably, I try to keep things as compatible as possible, but sometimes some changes impact old configurations.

thank you @set-soft! I am often surprised how quick you manage to implement new features and fix bugs. This is really great! Usually this means I can not keep up to it, and in most cases it does not matter. As my template is just a set of predefined config files all necessary documentation can be found in the kibot readme.
Sometimes I am not aware of breaking changes, so if you @Informaticore experience further issues here please feel free to modify these config files... they are a TEMPLATE, they are not meant to work for every body under very circumstances.