INTI-CMNB/KiBot #425: [HOW] KiCad 7.0.2

set-soft, 2023-04-17:
Lamentably KiCad 7.0.2 was released too early, with some very annoying bugs:

1. All power symbols will produce an ERC error about a missing simulation model. This ERC rule will default to IGNORE in 7.0.3 and KiCad won't check simulation models for power sources or for components marked as excluded from simulation. 
2. ERC dialog some times closes right after finishing the ERC. Not sure if this is already addressed.
3. The EDA_ITEM.Duplicate() member doesn't cast the returning value. So some code using ZONES fails. This is the case for KiKit panelize.
4. The code doesn't compile for the Debian 11 wxWidgets. Now fixed.

I'll add workarounds for most things, not sure if all. I'll keep this issue opened until 7.0.3 is released.

laurencedv, 2023-04-18:
Thanks for supporting in anycase

set-soft, 2023-05-30:
Ok, KiCad 7.0.5 is out, passes all the KiBot regression tests (just details in the generated aspect, be careful with projects started with KiCad 7 and justified text, it changed to KiCad 6 rules, so the justification will change).
I updated the docker images, so I'm closing this.

closed 2023-05-30 by set-soft