INTI-CMNB/KiBot #427: [HOW] to render via Blender - ERROR:Malformed component reference `3V3`

frippe75, 2023-04-20:
**What do you want to achieve?**
Creating a Blender 3D output

**Do you have some PCB/Schematic to use as example?**
no example

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
kibot:
  version: 1

globals:
  # Filters for KiBot warnings
  filters:
    - filter: Ignore PcbDraw missing components
      number: 103
    - filter: Ignore missing KiCad config, not in docker images
      number: 8
    - filter: Ignore missing KiCad 3D models, not in docker images
      number: 10
    - filter: Ignore missing KiCad config from KiAuto, not in docker images
      number: 58
      regex: kicad_common.json
  # This will undo the `set_text_variables` when we run it locally
  restore_project: true
  # We will use a value externally formatted
  time_reformat: false

outputs:
  - name: 3D_top_view_HQ
    comment: "3D render from top (High Quality)"
    type: blender_export
    dir: 3D
    options:
      render_options:
        transparent_background: true
        samples: 20
      point_of_view:
        rotate_x: 30
        rotate_z: -20
      outputs:
        - type: blender
        - type: render

```

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a desktop system? Fedora 36 and KiCAD 7

**Additional context**
Add any other context about your setup/project.

```
 '3D render from top (High Quality)' (3D_top_view_HQ) [blender_export]
DEBUG:Current text variables: {} (kibot.gs - gs.py:212)
DEBUG:Output destination: /home/frta/3D (kibot - kiplot.py:369)
DEBUG: - Creating VRML ... (kibot - out_blender_export.py:378)
DEBUG:Starting tool check for KiCad Automation tools (kibot - dep_downloader.py:826)
DEBUG:- Returning `/home/frta/.local/bin/pcbnew_do` (kibot - dep_downloader.py:833)
DEBUG:Loading sheet from /home/frta/kicad/LedBee/ESP32-LedBee-Controller/ESP32-S3-LedBee-Controller_v2.kicad_sch (kibot - v6_sch.py:1855)
DEBUG:Loading sheet from /home/frta/kicad/LedBee/ESP32-LedBee-Controller/USB-C_v1.kicad_sch (kibot - v6_sch.py:1855)
DEBUG:Loading sheet from /home/frta/kicad/LedBee/ESP32-LedBee-Controller/MP2460.kicad_sch (kibot - v6_sch.py:1855)
ERROR:Trace stack: (kibot - error.py:33)
  File "/home/frta/.local/lib/python3.10/site-packages/kibot/kiplot.py", line 257, in load_any_sch
    sch.load(file, project)
  File "/home/frta/.local/lib/python3.10/site-packages/kibot/kicad/v6_sch.py", line 2016, in load
    comp.set_ref(s.reference)
  File "/home/frta/.local/lib/python3.10/site-packages/kibot/kicad/v6_sch.py", line 947, in set_ref
    self.split_ref()
  File "/home/frta/.local/lib/python3.10/site-packages/kibot/kicad/v5_sch.py", line 1047, in split_ref
    raise SchError('Malformed component reference `{}`'.format(self.ref))
ERROR:While loading `/home/frta/kicad/LedBee/ESP32-LedBee-Controller/ESP32-S3-LedBee-Controller_v2.kicad_sch` (kibot - kiplot.py:269)
ERROR:Malformed component reference `3V3` (kibot - kiplot.py:270)
```

set-soft, 2023-04-21:
Hi @frippe75 !

I think the error is clear enough, but may be not. It says "Malformed component reference". This is because you are using "3V3" as the **reference** for a power element, this is the **value** (3.3 V), the reference for a power element should be something like "#PWRn" with "n" a number.

Use one or more letters followed by one or more numbers for references. The # is used for special stuff that isn't really a physical component, power sources are the most common. Note that KiCad relies on "#PWR" references for various things.

closed 2023-04-21 by set-soft

set-soft, 2023-04-21:
You can always use a KiCad demo for testing purposes.
For kit-dev-coldfire-xilinx_5213 I used:

```
kibot:
  version: 1

outputs:
  - name: 3D_top_view_HQ
    comment: "3D render from top (High Quality)"
    type: blender_export
    dir: 3D
    options:
      render_options:
        transparent_background: true
        samples: 20
      light:
        pos_x: -size*1.67
        pos_y: size*1.67
        pos_z: size*2.5
      point_of_view:
        rotate_x: 30
        rotate_z: -20
      outputs:
        - type: blender
        - type: render
```

frippe75, 2023-04-21:
@set-soft thanks for the response! Not sure I'm 100% onboard on the solution here. Would be nice to be able to turn on all those enforcements to quickly get to a rendering to determine if kibot is worth investing more time into.

I meant turn off the checking and simply just create the render. I used exactly that exemple to configure outputting of a blender rendering.

set-soft, 2023-04-21:
Try moving away the schematics, work in a directory containing just the PCB and the KiBot configuration. The schematic is optional for 3D render.