INTI-CMNB/KiBot #428: [HOW]to set environment variable for user §D modell repository

MarkusSBS, 2023-04-26:
**What do you want to achieve?**
Hello 

I want to integrate a reposirory with User 3D modells into a kibot CI.

At the Kicad side we have at the moment a workflow with a git repository with all 3d modell we use additional to the kicad internal modells.
to integratethem to kicad we defined a environment variable  KICAD_USER_3DMODELS_DIR.

this is used inside the footprint lib  for example:${KICAD_USER_3DMODELS_DIR}/EATON_1245HC.STEP	

the integration of the git is not a big problem with:

before_script:
  - git -c http.sslVerify=false clone https://gitlab-ci-token:${CI_JOB_TOKEN}@gitlab.intra ..........

At the moment i am not totaly sure how to integrate a self defined enviroment var into kibot 

Global/environment looks like that it just overwrites an predefined value? 

```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? `kicad_auto:dev_k6`,

set-soft, 2023-04-26:
Hallo @MarkusSBS !

> Global/environment looks like that it just overwrites an predefined value?

Yes, you are right.

You can just export an environment variable, you can do it after running `git clone`. Something like:

```
export KICAD_USER_3DMODELS_DIR=PATH_TO_3D_MODELS
```

To make it more robust you can define it as an absolute path, let say you are cloning the models in *SOME_DIR*, then you can:

```
export KICAD_USER_3DMODELS_DIR=`pwd`/SOME_DIR
```

MarkusSBS, 2023-04-26:
i tested:

 
- git -c http.sslVerify=false clone https://gitlab-ci-token:${CI_JOB_TOKEN}@/hardware/sbs-kicad-3dmodels.git /root/.local/share/kicad/6.0/3dmodels//
  - export KICAD_USER_3DMODELS_DIR=/root/.local/share/kicad/6.0/3dmodels//
  - ls /root/.local/share/kicad/6.0/3dmodels//
  - ls ${KICAD_USER_3DMODELS_DIR}

but still i get the error:

set-soft, 2023-04-26:
Are you running KiBot in the same shell where KICAD_USER_3DMODELS_DIR was defined @MarkusSBS ?

I.e. next like after the `ls ${KICAD_USER_3DMODELS_DIR}`

set-soft, 2023-04-26:
Ok, this should be fixed/changed.

Try this:
- Open your project in the KiCad GUI
- Open the PCB Editor
- Choose File menu, Board Setup ... option, Text Variables
- Add an entry for "KICAD_USER_3DMODELS_DIR" defined as "/root/.local/share/kicad/6.0/3dmodels/"
- Confirm with Ok
- Exit KiCad
- Check the variable was added (using `git diff`)
- Commit and push the kicad_pro

Now the text variable should be expanded as a path.

MarkusSBS, 2023-04-27:
`set_text_variables:` 
- name: 'KICAD_USER_3DMODELS_DIR'
      text:  '/root/.local/share/kicad/6.0/3dmodels/'
 worked

set-soft, 2023-04-27:
I leave it open so other mechanisms are contemplated.

set-soft, 2023-05-02:
Patch 4fbd69ac3bbb15aa9f40769d952f11d4603227e8 allows defining environment variables in the global section.
Patch 4e72af9761b9e781e31450728da279500ac26182 allows expading environment variables in ${VAR} definitions.

closed 2023-05-02 by set-soft