INTI-CMNB/KiBot #429: [HOW] Add components marked with "Exclude from bill of materials" to position output dnf_filter

Swij, 2023-05-04:
**What do you want to achieve?**
Components that are marked with "Exclude from bill of materials" in the schematic should not be included in the `type: position` output.

I have added the option `dnf_filter: _kibom_dnf`, but this filter only removes components with "dnf" in the `_kibom_dnf_FIELD`.

Is there a filter for components with the "Exclude from bill of materials" marking?

**Do you have some PCB/Schematic to use as example?**
Just add a resistor to the schematic and check the "Exclude from bill of materials" box in its properties.

**Do you have some configuration file (.kibot.yaml) that you are using?**
```yaml
  - name: 'position_example'
    type: 'position'
    options:
      dnf_filter: _kibom_dnf
```

**Environment (please complete the following information):**
I'm running KiBot both locally (Debian) and in a CI/CD workflow on GitLab and GitHub. All use the `kicad7_auto_full:latest` docker image.

set-soft, 2023-05-04:
Hi @Swij !
Currently we don't have such an option. I think it could be implemented.
Note that you are using the wrong flag, in order to exclude components from the position file you must use the PCBNew option that explicitly asks to remove them. Which now I found doesn't work correctly when only SMD components are selected.

set-soft, 2023-05-05:
Hi @Swij !

The above patches fixes the above mentioned problem with *Exclude from position files* and adds two new options to the `generic` filter to filter using the *Exclude from bill of materials* and *Exclude from board* flags.

Please check if you can use them.

closed 2023-05-05 by set-soft

Swij, 2023-05-05:
Wow, thanks! That was quick :) 

Looks like it's working now with the dev image.

commit a1f78c29 (2023-05-05): [Filters][Added] New `generic` filter options

- `exclude_not_in_bom` and `exclude_not_on_board` to use KiCad 6+ flags

Closes #429