INTI-CMNB/KiBot #432: [FEATURE] Allow not saving (or reverting) set_text_variables changes to file

matthijskooijman, 2023-05-11:
**Is your feature request related to a problem? Please describe.**
I can use set_text_variables to set some variables, in my case version numbers / commit hashes derived from git. This allows me to, during development, export a PDF schematic with kibot that specifies exactly the git version from which it was generated.

However currently, `set_text_variables` makes changes to the project file on disk, leaving those changes in my working copy after the kibot run. This poses a risk that those changes will be accidentally committed to the git repo, with values that are no longer correct.

**Describe the solution you'd like**
It would be nice if the values set by `set_text_variables` would be applied in memory only, rather than on disk. Or, if that is not feasible, that they are reverted after kibot completes.

Rather than reverting the individual text variable changes, I can also imagine that kibot just makes a backup of the project file before running and restores that afterwards. This could be applied to all schematic and board files too, and controlled by some global `revert_all_changes` config value, and then this issue would be solved generically for all operations that need to modify files. 

**Describe alternatives you've considered**
 - Omitting the text variable update, but then generated PDFs are unclear about what version they contain.
 - Adding the version info in the `pcb_print` / `pdf_sch_print` output only, but those support modifying the `title` value only, not the arbitrary text variables that I'm using to display version numbers on my custom drawing sheet.

set-soft, 2023-05-11:
Hi @matthijskooijman !

> Rather than reverting the individual text variable changes, I can also imagine that kibot just makes a backup of the project file before running and restores that afterwards. This could be applied [...]

Try using the global variable `restore_project`.
KiBot won't save changes to the schematic.
Changes to the PCB are saved only when you ask for it (i.e. when refilling zones)

So I think  `restore_project` is what you need.

matthijskooijman, 2023-05-11:
Hah, that's pretty much what I proposed and exactly what I need. Must have overlooked when going through the (impressively long and complete) documentation before. Awesome that it is already implemented, that gives me more options :-)

I've just tested this and it works perfectly! Thanks for the super-quick response!

closed 2023-05-11 by matthijskooijman

set-soft, 2023-05-11:
I read your other reports, but right now I'm trying to make things work with the current git code of KiCad, 7.0.3 will be released soon, and I want to avoid the surprises I found in 7.0.2 ;-)

matthijskooijman, 2023-05-11:
No worries, I'm not blocked by either of them, just wanted to share some feedback.