INTI-CMNB/KiBot #435: [FEATURE] Allow setting groups in the outputs section too

matthijskooijman, 2023-05-11:
Fill as much as you can, remove the rest

**Is your feature request related to a problem? Please describe.**
I have a number of outputs defined, which need to run in different contexts. I.e. some need to run always, some only when generating for a variant, some only on releases, etc.

I started doing this by having different config files with subsets of the outputs, importing the config files with the outputs I need, but this isn't so nice (and if an output needs to be in multiple subsets, it gets messy quickly). Then I discovered you can define groups of outputs, which solves this problem more elegantly. However, the definition of groups is separate from the definition of outputs, so this needs duplicating the output names in the group section. More importantly, when adding a new output, it is easy to forget to add it to the relevant groups, because the groups are not defined along with the output itself.

**Describe the solution you'd like**
Adding a `groups: ['group1', 'group2']` configuration directive to all outputs, which adds the output to the groups listed. This would be functionally equivalent to listing the name of the output in the `groups` section under the named groups.

It is probably good to only allow this for groups that already exist (i.e. require defining the groups in the `groups` section as well, but with possibly empty output lists) to prevent a group name typo in the `outputs` section from going unnoticed.

set-soft, 2023-05-18:
Added, please check it works, I just did a very naive test.
You can also get a list of groups.

closed 2023-05-18 by set-soft

matthijskooijman, 2023-05-24:
I took this for a test drive, and the listing works well, but when you try to run kibot, it complains:
```
$ cat foo.kibot.yaml 
kibot:
  version: 1

groups:
 - name: a

outputs:
 - name: report
   type: report
   groups: a
$ kibot
ERROR:In section 'report' (report): Unknown option `groups`. Valid options: ['category', 'comment', 'dir', 'disable_run_by_default', 'extends', 'name', 'options', 'output_id', 'priority', 'run_by_default', 'type'] (kibot.gs - gs.py:732)
```

I've traced the problem to be the pre-parsing of the groups attribute, which replaces the `Optionable` instance in the `groups` attribute with the actual value (a list), which is then dropped from the list of available attributes by `Optionable.get_attrs_for()` leading to the above error.

Here's a path that works, but I'm not sure if this is a clean solution:

```diff
--- a/kibot/config_reader.py
+++ b/kibot/config_reader.py
@@ -155,6 +155,8 @@ def _parse_output(self, o_tree):
         o_out.groups = o_tree.get('groups', [])
         if isinstance(o_out.groups, str):
             o_out.groups = [o_out.groups]
+        # Remove to prevent parsing it again
+        o_tree.pop('groups', None)
         return o_out
 
     def _parse_outputs(self, v):
```

set-soft, 2023-05-24:
Ok, I fixed it. I uses *_groups* attribute to keep the pre-parsed groups and let *groups* just to get the validations.

commit ef04782a (2023-05-18): [Groups][Added] outputs can request to be added to one or more groups

- Also list groups
- Allow empty groups so the user can catch typos

Closes #435