INTI-CMNB/KiBot #437: [BUG]

matthijskooijman, 2023-05-13:
**Describe the bug**
When making a schematic diff of a multi-sheet schematic, with `new: current`, the sheet names of the new version are modified, producing unintended changes in the diff.

**To Reproduce**
I don't have time right now to produce a reduced example (and my schematic is not public), but I expect it happens with any multi-sheet schematic. This is the kibot output config I'm using:

```
- name: diff_sch_uncommitted
  comment:  Schematic diff of uncommited changes against current HEAD
  type: diff
  dir: Diff
  layers: all
  run_by_default: false
  options:
    old: HEAD
    old_type: git
    only_different: true
    pcb: false
    cache_dir: /tmp/kicad-diff-cache
```

**Expected behavior**
No difference should be generated when there is none.

**Screenshots**
This shows that the filename is changed:

[image]

The diff is not superclear, but looking at the exported PNG in the cache dir shows the green version more clearly:

[image]

The file in my schematic is called `EEPROM.kicad_sch`, so kibot changed the filename.

**Environment:**
Ubuntu 22.10, Kibot 1.6.2 installed with pip. KiDiff 2.4.7 installed from Debian repo.

**Additional context**
It seems this is caused by Kibot writing back the in-memory version of the schematic to file, where it (I think) uses an alternative "flat" filename, adding numbers (and flattening the directory structure). I'm not sure why this is needed, but it seems problematic for the diff usecase.

I've identified the code involved (see below), but I can't offer advice on a fix.

If I apply the below patch (which is not a fix, just a test), the filenames are kept the same and no longer show up in the diff.

```diff
--- /home/matthijs/.local/lib/python3.10/site-packages/kibot/kicad/v6_sch.py    2023-05-13 12:11:24.614271617 +0200
+++ /tmp/v6_sch.py      2023-05-13 12:11:21.970287553 +0200
@@ -1540,8 +1540,6 @@
                     _symbol('uuid', [Symbol(self.uuid)]), Sep()])
         for p in self.properties:
             change_file = cross and p.name in SHEET_FILE
-            if change_file:
-                p.value = self.flat_file
             data.extend([p.write(), Sep()])
             if change_file:
                 p.value = self.file
@@ -1836,7 +1834,7 @@
             saved.add(fname)
         for sch in self.sheets:
             if sch.sch:
-                sch.sch.save(sch.flat_file if cross else sch.file, dest_dir, base_sheet, saved)
+                sch.sch.save(sch.file, dest_dir, base_sheet, saved)
 
     def save_variant(self, dest_dir):
         fname = os.path.basename(self.fname)
```

matthijskooijman, 2023-05-13:
I've double-checked that this indeed caused by diffing with "new_type: current", since diffing with a commit just checks out that commit and then runs the diff, without loading/saving the schematic first.

e.g. this config produces an empty diff as expected:

```
- name: diff_sch_uncommitted
  comment:  Schematic diff of uncommited changes against current HEAD
  type: diff
  dir: Diff
  layers: all
  run_by_default: false
  options:
    old: HEAD^^
    old_type: git
    new: HEAD^
    new_type: git
    only_different: true
    pcb: false
    cache_dir: /tmp/kicad-diff-cache
```

set-soft, 2023-05-13:
Hi @matthijskooijman !
Please try using the current **dev** branch.
This is a known side effect of the variants, that was already discussed in the issues.
The new code applies this change of names only when strictly necessary.
This change is needed to overcome a huge KiCad limitation in the way shared pages are handled.

matthijskooijman, 2023-05-13:
Ah, I missed that. Just tested the dev branch, it is indeed fix. I guess this can be closed?

set-soft, 2023-05-13:
Ok, closing it

closed 2023-05-13 by set-soft