INTI-CMNB/KiBot #438: [BUG] Schematic diff against current with some project settings shows unintended changes

matthijskooijman, 2023-05-13:
**Describe the bug**
When generating a diff with `new_type: current`, schematic files are written to a temporary directory for diffing, but the project file is not written. If the project has settings that influence schematic rendering (such as `text_offset_ratio`), this causes differences that do not really exist.

I already checked the current dev branch, probably still happens there.

**To Reproduce**
 - Clone this project: https://github.com/matthijskooijman/kibot-bug-example-project
 - Run `kibot` to diff HEAD against current
 - See that this produces a diff when it should not:

[image]

(look carefully: there is some red/green due to text scale differences)

The example project was created as follows:
 - Create a new project.
 - Open shematic, create hierarchical sheet.
 - Open subsheet, add a hierarchical label.
 - Return to main sheet, import pins from subsheet.
 - Open schematic setup, change label offset ration from 15% to 25%.

**Expected behavior**
No differences should be produced.

**Environment:**
Ubuntu 22.10, Kibot 1.6.3 (dev branch 7e41c722) installed with pip. KiDiff 2.4.7 installed from Debian repo.

**Additional context**
The cause seems to be that when using `current` for a diff, the schematic files are written to a temp directory, but the project file is not. This means that rendering for a commit happens in the main directory, where the project file is present, but rendering for "current" happens in a new directory without the project file (so with default settings).

A fix would probably to just copy the project file into the temp dir. Alternatively, instead of writing into temporary directory, you could write to the main directory and rely on git to revert the changes, but that can probably fail if some files are not added to git.

matthijskooijman, 2023-05-13:
I tried a quick fix (since I now had a working git version anyway), not sure if this is completely the right fix, but it does work. See #439.

set-soft, 2023-05-16:
This patch should fix it.
I'll explain more in #439 
Please check it doesn't break another thing.

closed 2023-05-16 by set-soft

matthijskooijman, 2023-05-17:
I did some testing with current git dev, which seems to work as expected. It fixes this problem both in my minimal test case and in my original project. I have not yet tested this as part of my full git workflow, because there is no docker image available for this version yet (maybe because of the test failure?).

I did notice this commit caused a test failure, and had a look at that, but couldn't quickly make the test scripts run locally, so I didn't investigate closely. I also saw you already triggered a new test workflow this morning, so I suspect you're already onto this :-)

set-soft, 2023-05-17:
> I did some testing with current git dev, which seems to work as expected. It fixes this problem both in my minimal test case and in my original project. I have not yet tested this as part of my full [...]

Yes, `dev` images are created only if the code passes the tests. They are over than 500 tests, run for 3 KiCad versions, so it must pass over than 1500 tests. This ensures some level of confidence, we need much more tests in fact.

> I did notice this commit caused a test failure, and had a look at that, but couldn't quickly make the test scripts run locally, so I didn't investigate closely. I also saw you already triggered a new test workflow this morning, so I suspect you're already onto this :-)

A cron job triggers a test using KiCad 7.99. But yes I'm looking the source of the discrepancy in a diff test.