INTI-CMNB/KiBot #444: [BUG] Missing parts in JLCPCB BOM and CPL production files

twtb, 2023-05-19:
Fill as much as you can, remove the rest

**Describe the bug**
Tried to use KiBot to generate fabrication files for JLCPCB, but some THT files were missing. It looks like only SMD parts are in the BOM and CPL.
The part that I was trying to use: https://jlcpcb.com/partdetail/Shenzhen_ZhihaoElec-FJ5463AH/C122356.

**To Reproduce**
Config file that uses the following template:
```
import:
  - file: JLCPCB

```
**Expected behavior**
All parts are in the production files.

**Screenshots**
-

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? GitHub
- In a GitHub action? Which tag are you using? v2_k7

**Additional context**
-

set-soft, 2023-05-20:
This template is for SMD parts, the default files you'll ask for pick & place assembly.

You can create a customized JLCPCB including all the components you want. Just use the *template* as a base, here is a [link](https://github.com/INTI-CMNB/KiBot/blob/master/kibot/resources/config_templates/JLCPCB.kibot.yaml). Note they are *templates* with common choices.

I'll consider adding templates with THT components so people can choose.

closed 2023-05-20 by set-soft

set-soft, 2023-05-20:
In my experience the cost of soldering THT components is much higher than just SMD. Plus the fact that anybody can solder THT, but not always SMD, in particular below 0603 size.
This is why this is the default. I realize this is not the only option,so I'll add another template.

twtb, 2023-05-22:
> In my experience the cost of soldering THT components is much higher than just SMD. Plus the fact that anybody can solder THT, but not always SMD, in particular below 0603 size. This is why this is the default. I realize this is not the only option,so I'll add another template.

In my case it was actually cheaper to let JLCPCB solder a single additional part instead of ordering the part myself and pay extra shipping costs. The only problem is that they will not place it as it was not in the BOM :(. It's very confusing as the symbol has an LCSC code assigned in the schematic.

set-soft, 2023-06-01:
Now we can do:

```yaml
kibot:
  version: 1

import:
  - file: JLCPCB_with_THT
```