INTI-CMNB/KiBot #445: Render_3D without board body

PurpleLobster, 2023-05-25:
Is there a possibility to render_3d without the board body so the inner layers become visible? 
KiCad's 3d_viewer has a property called _show_board_body_  (part of the 3d_viewer.json config file). 
Unfortunately I cannot set this value to _False_ using the KiBot .yaml file.  

When running KiBot I get the following warning:
`WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:351)`
Could this mean the 3d_viewer.json config file is not used?

Reading other Issues I noticed 'blender_export' is recommended above render_3d. Two reasons are blocking me taking that approach:
1. PCB2Blender doesn't seem to export the inner layers
2. When running the example blender_export, only a .pcb3d file is exported. I can import this in Blender but blender_export properties like point_of_view are lost, somehow the rendering doesn't start.

set-soft, 2023-05-29:
Hi @PurpleLobster !

> Is there a possibility to render_3d without the board body so the inner layers become visible? 

I never tried it.

> KiCad's 3d_viewer has a property called _show_board_body_ (part of the 3d_viewer.json config file). Unfortunately I cannot set this value to _False_ using the KiBot .yaml file.

Is this enough to get what you want using KiCad?

> When running KiBot I get the following warning: `WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:351)` Could this mean the 3d_viewer.json config file is not used?

Nope. Where and how are you running KiBot? (local machine? CI/CD? etc.)

> Reading other Issues I noticed 'blender_export' is recommended above render_3d. Two reasons are blocking me taking that approach:
> 
> 1. PCB2Blender doesn't seem to export the inner layers

KiCad exports a VRML as a mechanism to export a 3D surface representing the PCB ... inner layers aren't part of the surface.

> 2. When running the example blender_export, only a .pcb3d file is exported. I can import this in Blender but blender_export properties like point_of_view are lost, somehow the rendering doesn't start.

Please be more specific, which example?
Also try running it with debug information (*-vvvv*), perhaps Blender is complaining about something.

set-soft, 2023-05-29:
BTW: Why do you want to get it on 3D? This is a normal thing you do for 2D using *pcb_print*

PurpleLobster, 2023-06-05:
Thank you for your suggestions @set-soft !

>Is this enough to get what you want using KiCad?

For now, yes. In the future Blender would be prefered but I assume that takes a while to figure out because the inner layers seem not supported.

>Nope. Where and how are you running KiBot? (local machine? CI/CD? etc.)

Local machine, I run 'kibot' in the Kicad project dir which includes the kibot .yaml file.

>Please be more specific, which example? Also try running it with debug information (_-vvvv_), perhaps Blender is complaining about something.

The basic blender export example created by running _kibot --quick-start_
Running with debug information was a very good suggestion, thx! 

_Blender 3.5.1 (hash e1ccd9d4a1d3 built 2023-04-24 23:31:15)
Importing PCB3D file ...
Error: Python: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/kibot/blender_scripts/blender_export.py", line 386, in <module>
    main()
  File "/usr/lib/python3/dist-packages/kibot/blender_scripts/blender_export.py", line 354, in main
    bpy.ops.pcb2blender.import_pcb3d(filepath=PCB3D_file, pcb_material=args.pcb_material,
  File "~/tools/blender-3.5.1-linux-x64/3.5/scripts/modules/bpy/ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "center_pcb" unrecognized_

I can't find any notion on the keyword "center_pcb".

>BTW: Why do you want to get it on 3D? This is a normal thing you do for 2D using pcb_print

To create nice visuals of the stackup used. Not for production.

PurpleLobster, 2023-06-05:
Setting the board color to fully transparent seems a step forward.
[image]

However, the layer colors are plotted in 'realistic_mode'. Is there an option to set realistic_mode to False?
My goal is to get something like this:
[image]

closed 2023-06-12 by set-soft

PurpleLobster, 2023-06-13:
Awesome @set-soft ! It seems to work well.

Although the the pcbnew_do section of the automation process now operates 2-5x slower than before but this might have to to with my setup. 
I checked out both KiBot and KiAuto repositories. Adding the pcbnew_do alias pointing to the executable in /src didn't work, KiBot ran the previous installed /usr/bin/pcbnew_do instead. Replacing the symbolic link worked out.

When I now run 'kibot' (the installed version 1.6.2), it seems slower as before as well, so I don't think it has anything to to with the new features.

Thanks a lot for the added features! You make my day :)