INTI-CMNB/KiBot #446: Errors with zipping four layer boards

tlalexander, 2023-05-26:
I have encountered an odd issue, and have written a fix. However it surprises me that this should even be possible.

I have a four layer board and I am just beginning to use it with KiBot. It succeeds at generating gerbers, but weirdly it would fail when trying to zip them. What was specifically weird is that the files are there, but the zipping code tries to come up with file names on its own, and a bug in the code that determines the extension meant that it could not find a file it was looking for and it throws an error.

Now, it knows the directory and the filename prefix, so I have no idea why it is even trying to guess the extension when it could just search the directory for a file with that prefix. In any case, the bug occurs because inner layers are not handled properly.

You can see in this line:
https://github.com/INTI-CMNB/KiBot/blob/eb6c2140f218f88222cfc5878f322c3a00e8115a/kibot/layer.py#L308

That any layer in Layer.DEFAULT_LAYER_NAMES is gets self._is_inner set to false. But on this line:
https://github.com/INTI-CMNB/KiBot/blob/eb6c2140f218f88222cfc5878f322c3a00e8115a/kibot/layer.py#L350

Inner layers do get added to Layer.DEFAULT_LAYER_NAMES. So it is wrong to always set self._is_inner to false.

This leads to a bug where a file with the In1_Cu prefix will be given the gbr extension, rather than the correct .g2 extension. The whole error I got was:
`ERROR:Unable to generate `/home/taylor/pcb/rp2040-motor-controller/JLCPCB/RP2040_motor-In1_Cu.gbr` from 'Gerbers compatible with JLCPCB' (JLCPCB_gerbers) [gerber] (kibot - out_compress.py:173)`

I fixed this in my copy of layer.py by changing the _is_inner assignment On L308 to the following:
```
            if self.layer[:2] == "In":
                self._is_inner = True
            else:
                self._is_inner = False
```
It feels like a hack as I would rather not do string matching, but it demonstrates the bug.

Once I saved that and ran kibot again, the error was gone and zipping completed successfully. Also all of my print debugging code I had added to other files confirmed that the filename was now correct during processing.

This should be repeatable with any four layer board, but I will post my board and yaml file soon if that is helpful.

set-soft, 2023-05-29:
Hi @tlalexander !

> Now, it knows the directory and the filename prefix, so I have no idea why it is even trying to guess the extension when it could just search the directory for a file with that prefix. In any case, the bug occurs because inner layers are not handled properly.

KiBot can't assume you are a meticulous person and you won't generate two gerber styles in the same directory, and then pretend to zip them in separated files. So putting anything from the directory isn't possible. Also imagine you generate some gerbers, and then you change your mind and generate another without deleting the previous, you won't want to get both in the zip file. And there are more things that makes impossible to go and look for what we got instead of *knowing* what we got. 

Thanks for looking at the code. The problem was exactly what you describe. I added a regression test, the extensions were checked, but the test didn't try to generate a compressed archive, now it does it.

closed 2023-05-29 by set-soft

tlalexander, 2023-05-30:
That makes sense. Thank you!

commit c87a37ea (2023-05-29): [Compress][Gerber][Fixed] Problems with inner layers

- When using legacy protel extensions

Fixes #446