INTI-CMNB/KiBot #450: [HOW] Generate STEP with DNP components

montirafa, 2023-06-13:
Generate STEP file with the components marked with "Do not Populate". I didn't find the option for this configuration and my output file needs to have all components.

set-soft, 2023-06-14:
Hi @montirafa !

Do you mean components marked with "Do Not Populate" in KiCad itself? If this is the case I'll recommend creating the STEP file using a separated KiBot config file that defines `kicad_dnp_applied: false` in the `global` section.

If you are marking the components using KiBot filters, just create a special filter and/or variant that doesn't exclude the components and use it just for the STEP generation.  In fact various examples are defining a dummy variant called default that doesn't exclude anything. You can get 3D renders and BoMs for all components, even DNPs.

closed 2023-06-14 by set-soft

montirafa, 2023-09-11:
Hi @set-soft, I used the first suggestion, which worked in the past. But, I ran today and it is not working anymore. I saw that has a new config _kicad_dnp_applies_to_3D_ but it doesn't work as well. Can we reopen the issue?

set-soft, 2023-09-12:
Hi @montirafa !
Please provide more information (KiCad, KiBot, etc. versions, environment) and a small example of KiCad files and KiBot project where the problem can be reproduced.