INTI-CMNB/KiBot #451: [BUG] Scaling of images in schematics fails when doing variant exports

N0cynym, 2023-06-15:
**Describe the bug**
I realized that for some reason kibot does scale the images incorrectly if you do `pdf_sch_print` with a variant.

**Export without variant:**

[image]

**Export with variant:**

[image]

**To Reproduce**

[Test.zip](https://github.com/INTI-CMNB/KiBot/files/11753235/Test.zip)

**Kibot Config**

```
variants:
  - name: 'default'
    comment: 'Variant with default C1, C2 = 100 nF, C3 = 100 uF R1 = 100 Ohm, R2 = 100 Ohm'
    type: kibom
    file_id: _default
    variant: default
    pre_transform: '_var_rename'
    exclude_filter: _none
  - name: 'variant_1'
    comment: 'Variant with default C1, C2 = 220 nF, without C3 R1 = 0 Ohm, R2 = 50 Ohm'
    type: kibom
    file_id: _variant_1
    variant: variant_1
    pre_transform: '_var_rename'
    exclude_filter: _none
  - name: 'variant_2'
    comment: 'Variant with default C1, C2 = 10 nF, without C3 R1 = 50 Ohm, R2 = 0 Ohm'
    type: kibom
    file_id: _variant_2
    variant: variant_2
    pre_transform: '_var_rename'
    exclude_filter: _none

outputs:    
  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
```
KiBot Call

```
kibot -d Schematics/default/ 
         -s run_drc,run_erc 
             print_sch

kibot -g variant=variant_1
         -d Schematics/variant_1/ 
         -s run_drc,run_erc 
             print_sch
```

**Expected behavior**

Scaling images if doing variant export as scaling if export without variant.

**Environment :**
Where are you running KiBot:
- In a CI/CD workflow (`GitLab`)
- Using docker image (`kicad7_auto:dev`)

set-soft, 2023-06-15:
The scale should be stored now, please test it.

closed 2023-06-15 by set-soft

N0cynym, 2023-06-15:
I can confirm that this works and it is fixed! 👍

commit 16a265f6 (2023-06-15): [Schematic v6][Fixed] bitmap scale and lib_name not saved

Fixes #451