INTI-CMNB/KiBot #452: [BUG] "Missing symbol instance" on Windows

martinhansdk, 2023-06-19:
**Describe the bug**
When running on Windows 11, latest KiBot fails to read the schematic, it prints the error

    Missing IM23 symbol instance for `/e12141d3-ccc7-4a3a-848f-e09010e2c0be/dc5e73fd-3355-489b-be6f-5209a46d6939\17c365bb-6fbe-4606-92cd-dc5247ab27bd` (kibot - __init__.py:1475)

The schematic consists of reproduce.kicad.sch which instantiates a usb_power.sch, which instantiates Relay.sch The symbol IM23 is instantiated in Relay.sch

KiBot config file:

```
kibot:
  version: 1

outputs:
  - name: 'pdf_sch_print_example'
    comment: 'PDF schematic'
    type: 'pdf_sch_print'
    dir: 'output/schematic'
```

The same project on Ubuntu 22.04.2 LTS (under WSL2) works fine.

Full output:

```
C:\Temp\reproduce>kibot -e reproduce.kicad_sch -vv
DEBUG:KiBot 1.6.2 verbose level: 2 started on 2023-06-19 09:18:28.576326 (kibot - __init__.py:1434)
DEBUG:Detected KiCad v7.0.5 (7.0.5 70000050) (kibot - __init__.py:1434)
DEBUG:KiCad config path C:\Users\MartinHansOOONOAS\AppData\Roaming\kicad\7.0 (kibot - __init__.py:1434)
WARNING:(W106) Running on Windows, this is experimental, please report any problem (kibot - log.py:102)
DEBUG:Command line arguments:
['C:\\Users\\MartinHansOOONOAS\\Documents\\KiCad\\7.0\\3rdparty\\Python39\\Scripts\\kibot', '-e', 'reproduce.kicad_sch', '-vv'] (kibot - __init__.py:1434)
DEBUG:Command line parsed:
{'--banner': None,
 '--board-file': None,
 '--cli-order': False,
 '--copy-and-expand': False,
 '--copy-options': False,
 '--define': [],
 '--dont-stop': False,
 '--dry': False,
 '--example': False,
 '--global-redef': [],
 '--help': False,
 '--help-banners': False,
 '--help-dependencies': False,
 '--help-filters': False,
 '--help-global-options': False,
 '--help-list-outputs': False,
 '--help-output': None,
 '--help-outputs': False,
 '--help-preflights': False,
 '--help-variants': False,
 '--invert-sel': False,
 '--json': False,
 '--list': False,
 '--log': None,
 '--makefile': None,
 '--markdown': False,
 '--no-auto-download': False,
 '--no-priority': False,
 '--no-warn': None,
 '--out-dir': '.',
 '--plot-config': None,
 '--quick-start': False,
 '--quiet': False,
 '--schematic': 'reproduce.kicad_sch',
 '--skip-pre': None,
 '--start': '.',
 '--type': [],
[... 248 lines cut ...]
  File "C:\Users\MartinHansOOONOAS\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kibot\kicad\v6_sch.py", line 2001, in load
    obj = SchematicComponentV6.load(e, self.project, self)
  File "C:\Users\MartinHansOOONOAS\Documents\KiCad\7.0\3rdparty\Python39\site-packages\kibot\kicad\v6_sch.py", line 1085, in load
    raise SchError('Missing {} symbol instance for `{}`'.format(comp.name, parent.get_full_path()))
ERROR:While loading `C:\Temp\reproduce\reproduce.kicad_sch` (kibot - __init__.py:1475)
ERROR:Missing IM23 symbol instance for `/e12141d3-ccc7-4a3a-848f-e09010e2c0be/dc5e73fd-3355-489b-be6f-5209a46d6939\17c365bb-6fbe-4606-92cd-dc5247ab27bd` (kibot - __init__.py:1475)
```

**To Reproduce**

Extract the attached project [reproduce.zip](https://github.com/INTI-CMNB/KiBot/files/11785463/reproduce.zip), start a KiCad 7.0 command prompt, then run

  kibot -e reproduce.kicad_sch -vv  

in the project directory.

**Expected behavior**
KiBot generates schematic output.

**Environment (please complete the following information):**
Where are you running KiBot:
- On Windows 11 in a command prompt started by the Start Manu item `KiCad 7.0 Command Prompt`
- KiBot was installed via pip

- The project was created using KiCad 7.0.5

set-soft, 2023-07-06:
Hi @martinhansdk !
Did you have a chance to test it?

martinhansdk, 2023-07-06:
Hi @set-soft 
Thank you for fixing this bug! I am currently on vacation but I will try your fix as soon as I'm back at my desk.

closed 2023-07-06 by martinhansdk

martinhansdk, 2023-07-24:
I have tested this and can confirm that is is fixed. Thank you!