INTI-CMNB/KiBot #454: Running DRC with library issues

montirafa, 2023-06-20:
**What do you want to achieve?**
I need to run the DRC for my project, but an error/warning is happening:

_WARNING:(lib_footprint_issues) The current configuration does not include the library 'user_components'.; Severity: warning
>     @(108.7000 mm, 115.4000 mm): Footprint X1 (pcbnew_do.kiauto.file_util - file_util.py:153)_

This warning is happening for all components of my project, but the libraries directory is correctly configured.

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a docker image? kicad_auto:ki7

set-soft, 2023-06-21:
I need much more information about the setup, project, configuration file, etc.

KiCad 7 introduced some checks that aren't useful. They can be used only for short periods of time. As soon as a new release of the libs is done you'll get tons of warnings. The solution is to just disable the tests that doesn't help. Not all the checks are useful for everybody. In fact they introduced some checks in 7.0.2 that wasn't specific for people wanting to simulate their designs (less than 0,01% because you won't get models for most stuff).

montirafa, 2023-06-21:
The configuration file and the log using -vvv are attached. I summarized the log with only 1 warning, but in fact there are 113 warnings (the same quantity of components on the board).

The filters didn't work, I think because is showing an error after the warnings: "ERROR:DRC violations: 113 (kibot.pre_run_drc - pre_run_drc.py:81)" and the outputs are not generated. When I used the filters this error also happens. 

[run drc.zip](https://github.com/INTI-CMNB/KiBot/files/11818249/run.drc.zip)

set-soft, 2023-06-22:
This isn't enough information. Do you have a public repo with this problem?
If the problematic lib isn't mapped in `fp-lib-table` Kicad will fail. 
Without knowing the name of the components, libs and how they are configured the error can be from multiple sources.
If you don't have a public repo just create one with a very small example that has this problem.
You are defining some text variables, but I don't know how they get defined for KiCad.

set-soft, 2023-07-06:
Hi @montirafa !
Do you have more information about it?

closed 2023-08-17 by set-soft