INTI-CMNB/KiBot #456: [BUG] Pick-n-Place issue with EuroCircuits

andersruneson, 2023-06-26:
**Describe the bug**
When uploading a CSV pick-n-place file to EuroCircuits it triggers their "Identify which column to use" dialog which adds manual work for each board ordered.

The issue seems to be a CSV with some columns having quotation marks and some don't. 

**To Reproduce**
Go to eurocuiruits.com, create account, do upload PCB-gerber-zip-file, upload the BOM and then upload a CSV pick-n-place-file generated by KiBot.

[image]

The generated pick-n-place-file looks like this:
```
Ref,Val,Package,PosX,PosY,Rot,Side
"C1","YAGEO_CC0402JRNPO9BN270","CAPC100X50X55L25N[0402]",144.1500,-27.7500,180.0000,top
"C2","YAGEO_CC1206KKX7R7BB225","CAPC320X160X125L50N[1206]",144.1000,-40.6000,270.0000,top
"C3","TAIYO_EMK107B7104KAHT","CAPC160X80X90L35N[0603]",142.5000,-27.4000,270.0000,top
```

If I add quotation marks " on each column like below, EuroCircuits can read the file fine:
```
Ref,Val,Package,PosX,PosY,Rot,Side
"C1","YAGEO_CC0402JRNPO9BN270","CAPC100X50X55L25N[0402]","144.1500","-27.7500","180.0000","top"
"C2","YAGEO_CC1206KKX7R7BB225","CAPC320X160X125L50N[1206]","144.1000","-40.6000","270.0000","top"
"C3","TAIYO_EMK107B7104KAHT","CAPC160X80X90L35N[0603]","142.5000","-27.4000","270.0000","top"
```

This is the output configuration we use for EuroCircuits:
```
# Same as our generic PNP, but without DNP components
- name: eurocircuits_CSV
  comment: Components position for Pick & Place
  type: position
  dir: 'Manufacturers/EuroCircuits'
  run_by_default: true
  options:
    output: '@EA@-@EArevLong@ EuroCircuits CPL.%x'
    format: CSV
    only_smd: false
    dnf_filter: 
    - no_x_in_DNP_field
    - _mechanical
    - elysics_extra_mech
    separate_files_for_front_and_back: false
```

**Expected behavior**
The import at EuroCircuits should happen automatically.

To be fair, the bug is at EuroCircuits, so I've reach out to them as well to get them to correct their parsing. But maybe add an option to have quotation marks on all columns?

**Screenshots**
See above.

**Environment (please complete the following information):**

**Additional context**

set-soft, 2023-06-26:
Hi @andersruneson !
I think the bug is in Eurocircuits, and what you need is a workaround for this bug.
KiBot is quoting the same values KiCad quoted when we implemented it. And what KiCad does (or did, never checked again) is quite logic. When you quote a number spreadsheets will interpret it as a string and you won't be able to do math with them, unless you convert them to numbers.
IMHO quoting numbers is a very bad idea, a waste of space and a potential source of problems.
That said: we could add an option to quote all.

andersruneson, 2023-06-26:
Yes, right, I found the issue when switching to KiBot because the script we used previously to generate pnp did not quote **any** fields, which can be a bit "dangerous" in case there is commas in the strings.

I also agree that the best place to fix this is at EuroCircuits, but let's keep this open until they reply. The other alternative is of course for me to write my own csv-to-csv converter. But in case this affects more than me it's nice to have it in KiBot.

Old script below, probably something we found on the internet, I just though that KiBot used the "same".
```
tableTop = []
tableBottom = []
for module in board.GetModules():
  attr = module.GetAttributes()
  if attr == MODULE_ATTR_NORMAL_INSERT or attr == MODULE_ATTR_NORMAL:
    (pos_x, pos_y) = module.GetPosition()
    ref = module.GetReference()
    ref = ref.replace(";", ",")
    val = module.GetValue()
    val = val.replace(";", ",")
    fp = str(module.GetFPID().GetLibItemName())
    fp = fp.replace(";", ",")
    x = "{0:.4f}".format((pos_x - origx)/1000000.00)
    y = "{0:.4f}".format((origy - pos_y)/1000000.00)
    rot = "{0:.4f}".format(module.GetOrientation()/10.0)
    side = 'top'
    if module.IsFlipped():
      side = 'bottom'
    data = {'Reference': ref,
            'Value': val,
            'Footprint': fp,
            'PosX': x,
            'PosY': y,
            'Rotation': rot,
            'Side': side
            }
    data = "{0[Reference]};{0[Value]};{0[Footprint]};{0[PosX]};{0[PosY]};{0[Rotation]};{0[Side]}".format(data)
    if side == 'top':
      tableTop.append(data)
    if side == 'bottom':
      tableBottom.append(data)
```

closed 2023-06-26 by set-soft

andersruneson, 2023-06-26:
Wow, that was fast! Thank you so much!

commit 7e6154d9 (2023-06-26): [Position][Added] quote_all option

- To quote all columns in the CSV output

Closes #456