INTI-CMNB/KiBot #460: [BUG] 3D model not found if path has spaces

mundodisco8, 2023-06-28:
Fill as much as you can, remove the rest

**Describe the bug**
Trying to run the example config on my project, where the 3D models have spaces in the filename and in the path to them. I already had a similar issue (although in a different environment, https://github.com/INTI-CMNB/KiBot/issues/366)

I attach the log as a file as it's quite long (-vvvv)

[output.log](https://github.com/INTI-CMNB/KiBot/files/11893326/output.log)

**To Reproduce**
This zip has the files I was using. Config is the example config, just with most of the stuff removed (only 3D models from blender)

```
kibot --schematic hardware/LED_Dimmer.kicad_sch --board hardware/LED_Dimmer.kicad_pcb
```

[Gruffalo_Dimmer.zip](https://github.com/INTI-CMNB/KiBot/files/11893300/Gruffalo_Dimmer.zip)

**Expected behavior**
Models with spaces in their names should have those spaces correctly escaped.

**Screenshots**
```
DEBUG:Missing 3D model file /LED_Dimmer/hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step (${KIPRJMOD}/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step) (kibot - out_base_3d.py:493)
WARNING:(W047) Missing 3D model for R2: `/LED_Dimmer/hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step` (kibot - out_base_3d.py:503)
DEBUG:- Expanded ${KIPRJMOD}/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step -> /LED_Dimmer/hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step (kibot - out_base_3d.py:83)
DEBUG:Missing 3D model file /LED_Dimmer/hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step (${KIPRJMOD}/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 - SMD.step) (kibot - out_base_3d.py:493)
```

**Environment (please complete the following information):**
- Using a docker image, kicad7_auto_full:latest

set-soft, 2023-06-28:
Hi @mundodisco8 !

The error says it can't find *Resistor - 0805 - SMD.step*, which isn't included in the zip you posted.

```
$ find . -name "Resistor*.step"
./hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0603 - 1.6x0.8mm - SMD  v1.step
./hardware/Libraries/Gruffalo_Dimmer.pretty/3D Models/Resistor - 0805 -3.2x1.6mm - SMD.step
```

Plus I don't understand where should I escape spaces when the test is done inside the code, not calling an external tool.
IMHO using spaces in file names is a very bad idea in general, but should work.

mundodisco8, 2023-06-28:
Sorry!

I'm a dumdum! You are totally right. I got distracted with the output. When I first got it, I copied and pasted one of the expanded names from a file that actually exists!

I'm closing this, couldn't see the forest for the trees

closed 2023-06-28 by mundodisco8