INTI-CMNB/KiBot #461: [HOW] kikit_present with just exported files

N0cynym, 2023-06-30:
**What do you want to achieve?**
I want to use the `kikit_present` export to create a project webpage.

My problem is that I still have exported via kibot all filles I want to add but `kikit_present` runs his own config in kibot and adds this exports.

If I say `mode: file` it uses my files but still have set the file type in the background.

```yaml
  - name: 'kikit_present'
    comment: 'Create a deployed Webpage for file presentation'
    type: kikit_present
    dir: .
    options:
      description: kicad-ci-tests.md
      name: 'kicad-ci-tests Version 1'
      boards:
        mode: file
        front_image: Production/Images/kicad-ci-tests-3D_blender_top.png
        back_image: Production/Images/kicad-ci-tests-3D_blender_bottom.png
        gerbers: Production/default/Production_default.zip
        pcb_file: Production/Documentation/Documentation.pdf
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow (`GitLab`)
- Using a docker image? (`kicad7_auto_full:dev `)

**Additional context**
Could we add additional files with any type to the `Available files` section?

set-soft, 2023-06-30:
Did you try `mode: local` like in this example: [kikit_present_local_1.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/master/tests/yaml_samples/kikit_present_local_1.kibot.yaml)

N0cynym, 2023-07-03:
This seems not to be helpful for my usecase because the board section should have multiple "boards" which should be the pick and place variants. This seems not to work as I want if I do export my variants in seral GitLab jobs.

If I creat for each variant a "new web presentation" it will work I guess. 

But the board section will still contain the same pre defined content.

I wish to add to the board section my own exports like:

- Schematics
- Documentation/Report
- PCB Layer PDF
- Gerbers
- Bill of Material
- Pick and Place data

Do you this we can add a feature which allows me to add my customized content?
Or a flag which let's me just use my `description:` mark down file which includes all my custom content and skips the board section?

N0cynym, 2023-07-03:
Ok just now I realized that `kikit_present` does not support markdown tables.

[image]

This seems to make it more difficult as I suspected...

My idea was to use the same markdown I use for my GitLab release pages:

[image]

and just ignore the `board` section ...

[image]

N0cynym, 2023-07-03:
Added [Issue#564](https://github.com/yaqwsx/KiKit/issues/564) to [KiKit](https://github.com/yaqwsx/KiKit).

set-soft, 2023-08-01:
Hi @N0cynym !

About KiKit: now the KiBot docker images should contain a KiKit that includes the patch to support markdown tables.

About presentation for variants: I think the best is to generate the variants using one set of config files, indicating which variant in the command line. Once the variants are generated create the presentation using a separated config that takes the files from the previous runs.

N0cynym, 2023-08-24:
@set-soft thank you very much I'm very happy with this feature and with your advices 🙏

closed 2023-08-24 by N0cynym