INTI-CMNB/KiBot #462: [BUG] Python exception after Upgrade from Debian 11 to 12

rainer38, 2023-07-03:
**Describe the bug**
After i upgraded my Debian distribution from bullseye (11) to bookworm (12) and reinstalled the needed Python packages, Kibot throws an exception when generation a POS file for assembling.

This is the thrown exception:

Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.3', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 520, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 527, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 517, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 418, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 197, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/usr/lib/python3/dist-packages/kibot/out_position.py", line 229, in run
    self.filter_pcb_components()
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 760, in filter_pcb_components
    self.sch_fields_to_pcb(GS.board, self.comps_hash)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 831, in sch_fields_to_pcb
    old_properties = m.GetProperties()
                     ^^^^^^^^^^^^^^^
AttributeError: 'FOOTPRINT' object has no attribute 'GetProperties'

**To Reproduce**
I can reproduce the problem with the example JLCPCB configuration from your website when using the option "variant: rotated". When i comment out the variant option it works.
This is my sample conf file: [export.kibot.zip](https://github.com/INTI-CMNB/KiBot/files/11935416/export.kibot.zip)

My command to execute it:
kibot -A -d /mnt/ramdisk -c /mnt/ramdisk/export.kibot.yaml jlcpcb_pos

**Environment (please complete the following information):**
I run KiBot v1.6.3 locally on my Debian 12 PC and use KiCAD 7 nightly.
The packages are installed from the KiBot repository,

set-soft, 2023-07-03:
> I run KiBot v1.6.3 locally on my Debian 12 PC and use KiCAD 7 nightly.
> The packages are installed from the KiBot repository,

Please try using a release. 
Explanation: The nightly tests are failing and honestly I don't have time to maintain them working, KiCad introduces breaking changes every week, some times more than once. So I end expending all my available time just trying to keep it working. KiCad stable isn't stable at all, the development team is cherry-picking changes from 8.x.x branch, which is highly unstable. The patches are applied without a minimal check. I tried to support them, but is impossible. Also: avoid using a fresh release, they cancelled various releases and KiBot will need time to adapt to the bugs introduced, most 7.0.x releases introduced bugs. 
For this reason I won't take issues for unreleased KiCad versions. Perhaps I'll do when they finally get somethings stable, but I don't see it happening in the near future.
If you can reproduce it using a release please open a new report. 
This problem seems to be related to some Python API breakage, they never pay attention to it and they rely on people submitting patches to fix it.

closed 2023-07-03 by set-soft

rainer38, 2023-07-04:
Just for your information.

The commits starting at 028f500cd9059e3d99567b582cfd76114f69ec94 and following in KiCAD breaks the functionality with KiBOT.

There were internal names in pcb objects changed to be consistent with the schematic objects.
I think starting with KiCAD v7.0.6 these changes will be deployed.

set-soft, 2023-07-06:
Hi @rainer38 !
Thanks for the commit hash.
As you can see this is a problem with KiCad changing API in a minor release, something common, and unneeded. This is the way KiCad is developed, paying more attention to cosmetic than fixing bugs.