INTI-CMNB/KiBot #463: [HOW] Image quality of `render_3d`

dmohns, 2023-07-12:
Hey,

**What do you want to achieve?**

I'm trying to render some 3D images as PNG. It works and KiBot generates some output file. However the output is of extremely low quality. I'm wondering how can I increase the DPI for example?

**Do you have some configuration file (.kibot.yaml) that you are using?**

You can attach it or paste the content in the following section:
```yaml
outputs:
- name: basic_ibom
  comment: Interactive HTML BoM
  type: ibom
  dir: Assembly
- name: basic_render_3d_30deg
  comment: 3D view from 30 degrees
  type: render_3d
  dir: 3D
  output_id: 30deg
  options:
    realistic: true
    rotate_x: 3
    rotate_z: -2
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? locally
- Using a docker image? (`kicad7_auto_full`)
- Using a desktop system? MacOS

set-soft, 2023-07-12:
Hi @dmohns !
For *render_3d* you'll get much better results using `ray_tracing: true`.
But if you are looking for high quality just switch to *blender_export* option.
Here is one [example](https://github.com/INTI-CMNB/kibot_variants_arduprog)
I guess you are looking for quality, not image size (DPI).

closed 2023-07-12 by set-soft

dmohns, 2023-07-13:
> Hi @dmohns ! For _render_3d_ you'll get much better results using `ray_tracing: true`. But if you are looking for high quality just switch to _blender_export_ option. Here is one [...]

Hey @set-soft thanks for the quick reply. I saw the possibility to use `blender_export`. But I'm specifically interested in generating images in a common image file format, preferably PNG. I don't think this is possible using `blender_export` without post-processing, right?

What I mean by quality is: The images generated by `render_3d` are very pixelated, it's hard to see anything on them. The DPI seems the same as KiCAD export, sorry for the confusion. It got better after playing around with the `width` and `height` settings.

set-soft, 2023-07-13:
> I don't think this is possible using `blender_export` without post-processing, right?

It has a rendered output! You can generate PNGs.

> What I mean by quality is: The images generated by `render_3d` are very pixelated, it's hard to see anything on them. The DPI seems the same as KiCAD export, sorry for the confusion. It got better after playing around with the `width` and `height` settings.

Remember to enable the ray tracing option in your KiBot config

dmohns, 2023-07-14:
> Remember to enable the ray tracing option in your KiBot config

Yeah, this already makes the image look much better.

> > I don't think this is possible using `blender_export` without post-processing, right?
> 
> It has a rendered output! You can generate PNGs.

Maybe it's just me, but the only output it is generating is a `pcb3d` file 🤔 This is my config

```
- name: 3D_top_view_HQ
  comment: "3D render from top (High Quality)"
  type: blender_export
  dir: 3D2
  options:
    render_options:
      transparent_background: true
      samples: 20
    point_of_view:
      rotate_x: 30
      rotate_z: -20
    outputs:
      - type: render
```

set-soft, 2023-08-04:
Hi @dmohns !
Your project is a little bit problematic, but the real problem is that you set the PCB thickness to 0. This confuses KiCad and generates an invalid WRL.

After fixing it, and downloading a couple of missing stuff that you have only in your PC, I got:

[image]

BTW: avoid using spaces in file names, it generates various problems

set-soft, 2023-08-04:
> This project was handed over to me by another developer. And I was struggling to find the missing files. Could you teach me where you found the missing files so quickly?

They are available at GrabCAD, I think you need to register in order to download them. Just ask Google to search the name of the missing step files. The _HW-111_RTCmodule_DS1307.STEP_ and _YAAJ_BluePill.step_ files will give perfect matches in Google and _PQ-26-20_6+6_tx.step_ will give partial match, but after uncompressing the file at GrabCAD you'll get the needed file.

I have them if you have troubles to get them, but the _YAAJ_BluePill.step_ model is 84 MB, so I think is simpler to just download it.

dmohns, 2023-08-04:
Thanks a lot! I will have a look to integrate these next week. 

This help tremendously 🙏