INTI-CMNB/KiBot #464: [HOW]KiCad Migration 6 => 7.0.5

MarkusSBS, 2023-07-12:
**What do you want to achieve?**
Migration of an "old" projekt to Kicad 7

I Am trying to migrate to kicad 7 but i get hundrets of DRC errors.
With the un edited config and drc_exclusions_workaround: true all ignored tests are exeuted and they fail.
If I remove the drc_exclusions_workaround: true option all exclusions lead to an error.
I do not use any other options for the DRC 

Did i miss a new workaround?

Where are you running KiBot:
- In a CI/CD workflow? GitLab

- Using a docker image?  ghcr.io/inti-cmnb/kicad7_auto_full:dev
- Using a desktop system? Windows

MarkusSBS, 2023-07-12:
[image]
[image]

MarkusSBS, 2023-07-17:
i think it was a problem with the project file deleting its erc and drc exclusions but i have to verify this.

MarkusSBS, 2023-07-17:
after save and close of the projekt the DRC and ERC section of the projekt file was just empty. At the moment i dont know if its an user error, KiCad Bug, or a git problem. I have to monitor this problem. editing the projekt file fixed it for now.

set-soft, 2023-07-18:
Ok, may be related to changes in the exclusions format.

closed 2023-07-18 by set-soft