INTI-CMNB/KiBot #470: [HOW]Integration of the variants: file_id: in User parameter and Ibom Titel

MarkusSBS, 2023-07-24:
Hello

Is it possible to  integrade the content of the file_id  into a user parameter for a variant.?

This would be usefull for a print on the PCB. The expansion of %V in file name works as expected.

and is it possible to include the variant into the Ibom title header?

set-soft, 2023-07-26:
Hallo @MarkusSBS !

> Is it possible to integrade the content of the file_id into a user parameter for a variant.?
> 
> This would be usefull for a print on the PCB. The expansion of %v in file name works as expected.

Not sure if I understand what you want: Do you want to expand %v in a text found in the PCB? (i.e. a silk screen text)
If you are using a global variant (i.e passing it from the command line) you can try defining a text variable (set_text_variables preflight) that contains "%v". It will define a text variable that will be stored in the project file. So then KiCad will be able to access this information.

> and is it possible to include the variant into the Ibom title header?

Do you mean the displayed top left name? I'm afraid this is the name of the KiCad project, not something we really control in a direct way. But it could be implemented because we use a copy of the project, so we could use a different name.

If this is what you want please open a feature request asking to allow specifying the project name used for a variant in the iBoM output.

MarkusSBS, 2023-07-26:
> Not sure if I understand what you want: Do you want to expand %v in a text found in the PCB? (i.e. a silk screen text)
> If you are using a global variant (i.e passing it from the command line) you [...]

thats what i want i was not aware that the %v works with every text not just the file name expansion 

> Do you mean the displayed top left name? I'm afraid this is the name of the KiCad project, not something we really control in a direct way. But it could be implemented because we use a copy of the project, so we could use a different name.

Yes i want to change the top left name to identify the version. like ` Fancyprojekt, WithoutUSB `

set-soft, 2023-07-31:
Hallo @MarkusSBS !
Try the above patch. Use something like `forced_name: '%f, %V'`

commit 68e0ccc9 (2023-07-31): [iBoM][Added] `forced_name` to control the displayed name

- To force the name displayed at the top left corner

Closes #470
closed 2023-07-31 by set-soft