INTI-CMNB/KiBot #471: [BUG]

nopeppermint, 2023-07-24:
**Describe the bug**
I have components in my Kicad Library which have in the "value" field  other strings which should be replaced but are not in Kibot.
For example: ${Value_C}_${Voltage} or ${Manufacturer_1_Part_Number} or ${Value_R}_${Tolerance}

Kibot generated some logs related to this:

```
ERROR:Unable to expand `Manufacturer_1_Part_Number` in `${Manufacturer_1_Part_Number}` (kibot - config.py:113)
WARNING:(W020) Malformed value: `${Value_C}_${Voltage}` (no match in C101) (kibot - units.py:224)
ERROR:Unable to expand `Value_L` in `${Value_L}_${Tolerance}` (kibot - config.py:113)
ERROR:Unable to expand `Tolerance` in `${Value_L}_${Tolerance}` (kibot - config.py:113)
ERROR:Unable to expand `Value_R` in `${Value_R}_${Tolerance}` (kibot - config.py:113)
```

**To Reproduce**

I tested a bigger project I can not share, with ```kibot --quick-start```

An example for a schematic symbol causing this issue:
[value_substitution_test.zip](https://github.com/INTI-CMNB/KiBot/files/12147352/value_substitution_test.zip)

**Expected behavior**
If I export the same BOM with Kicad ${Value_C}_${Voltage} is actually resolved to "1u_16V"

**Screenshots**
[image]

KiBot BOM:
[image]

KiCad BOM:
[image]

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`) no
- In a GitHub action? no
- Using a docker image? no
- Using a desktop system: in WSL ubuntu 22.04 on Win10

```
$kibot-check
KiBot installation checker

Core:
Linux: 5.15.90.1 (Linux MSW01-WS07 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux)
Python: 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
KiCad: 7.0.6-7.0.6~ubuntu22.04.1
Kibot: 1.6.3

Modules:
Colorama: 0.4.4
LXML: 4.8.0
Lark: 1.1.1
PyYAML: 5.4.1
QRCodeGen: Ok
Requests: 2.25.1
XLSXWriter: 3.0.2
Xvfbwrapper: Ok
markdown2: 2.4.1
mistune: 0.8.4
numpy: 1.21.5

Tools:
Bash: 5.1.16 (GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu))
Blender: 3.6.1 (Blender 3.6.1)
Ghostscript: 9.55.0
Git: 2.34.1 (git version 2.34.1)
ImageMagick: 6.9.11.60 (Version: ImageMagick 6.9.11-60 Q16 x86_64 2021-01-25 https://imagemagick.org)
Interactive HTML BoM: 2.6.0 (v2.6.0)
KiBoM: 1.9.0 (KiBOM Version: 1.9.0)
KiCad Automation tools (kiauto): 2.2.6 (pcbnew_do 2.2.6 - Copyright 2018-2023, INTI/Productize SPRL - License: Apache)
KiCad PCB/SCH Diff (kidiff): 2.4.7 (kicad-diff.py 2.4.7 - Copyright 2020-2023, INTI/Salvador E. Tropea - License:)
KiCost: 1.1.18 (KiCost v1.1.18)
KiKit: 1.3.0.7 (kikit, version 1.3.0-7)
OpenSCAD: 2021.1.0 (OpenSCAD version 2021.01)
Pandoc: 2.9.2.1 (pandoc 2.9.2.1)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.52.5 (rsvg-convert version 2.52.5)
Xvfb: Ok (xvfb-run)
```

set-soft, 2023-07-24:
Never saw somebody using such a feature. 
Will see about it the next week.

set-soft, 2023-07-31:
Hi @nopeppermint !
The above patch should add it. Let me know if it works for your project.

commit 055e8326 (2023-07-31): [BoM][Added] Support for ${field} expansion

Closes #471