INTI-CMNB/KiBot #473: [BUG] change in behavior of compress

MarkusSBS, 2023-07-26:
**Describe the bug**
After the switch to the Kicad7 train compress stops working if you use the source input and it  just generates an empty zip file.
For a another compress jop it just worked fine if the input is a another job output

This Workes just fine:

```
  - name: JLCPCB
    comment: ZIP file for JLCPCB
    type: compress
    dir: JLCPCB
    options:
      files:
        - from_output: JLCPCB_gerbers
          dest: /
        - from_output: JLCPCB_drill
          dest: /

```

This generates an empty file in /Fabrication/

```
  - name: 'archiv'
    comment: "ZIP archive"
    type: compress
    dir: ../Fabrication
    options:
      files:
        - source: ../Fabrication/dril/*
          dest: /
        - source: ../Fabrication/Gerbers/*
          dest: /
        - source: ../Fabrication/JLCPCB/*
          dest: /
        - source: ../Fabrication/Produktion/*
          dest: /
        - source: ../Docu/report/*.pdf
          dest: /
      output: "EMS_Produktion_%D-%T.%x"
```

If I execute an `ls `  in the script i can see that it is in the right direktory and its filled with the same data  as the resulting job Artefakts.
It worked with the K6 dev docker some weeks ago.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? ghcr.io/inti-cmnb/kicad7_auto_full:dev

set-soft, 2023-07-31:
Hallo @MarkusSBS !
Please check the debug output to determine when is the `archiv` output generated. Make sure it runs after generating the stuff you want to compress. The debug output will also inform which files are included.

set-soft, 2023-07-31:
Hallo @MarkusSBS !
I think the problem is that you are missing `from_cwd: true`. Because all your `ls` commands refer to `./`, but then you specify `-d Release` and the doc says:

- **`source`**: [string='*'] File names to add, wildcards allowed. Use ** for recursive match.
                            By default this pattern is applied to the output dir specified with `-d` command line option.
                            See the `from_cwd` and `from_output_dir` options.