INTI-CMNB/KiBot #474: [FEATURE] Specify optional layer for output

Kedarius, 2023-07-31:
**Is your feature request related to a problem? Please describe.**
I am creating a flex PCB. For that I have renamed one user layer to Stiffener. I am using one config file for all projects - and one automation too. 

**Describe the solution you'd like**
I would like to be able to specify that some layer is optional and KiBot should not error when it is not found in the board.

**Describe alternatives you've considered**
I have considered creating special config file for each project and/or some mechanism howto modify the global one with project specific settings. I find it more complicated and more error prone....

**Additional context**
```
The stack trace is here:
ERROR:Trace stack: (kibot.gs - gs.py:719)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 418, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_any_layer.py", line 331, in run
    self.options.run(output_dir, self.layers)
  File "/usr/lib/python3/dist-packages/kibot/out_any_layer.py", line 174, in run
    layers = Layer.solve(layers)
  File "/usr/lib/python3/dist-packages/kibot/layer.py", line 222, in solve
    layer._get_layer_id_from_name()
  File "/usr/lib/python3/dist-packages/kibot/layer.py", line 324, in _get_layer_id_from_name
    raise KiPlotConfigurationError("Unknown layer name: `{}`".format(self.layer))
ERROR:In section 'JLCPCB_gerbers' (gerber): Unknown layer name: `Stiffener` (kibot.gs - gs.py:733)

```

set-soft, 2023-08-01:
Hi @Kedarius !

I think what you want to achieve can be done using [definitions](https://github.com/INTI-CMNB/KiBot/tree/dev#doing-yaml-substitution-or-preprocessing).
Is this enough?

Kedarius, 2023-08-01:
Thanks for your reply. I did completely miss this feature in the read me. To be honest I find the one big readme file hard to navigate. However yes, it seems that I will be able to use this.  I was not able to do it as elegant as I would like but its mainly because me not knowing YAML good enough.
I will post it here for future readers.
So I changed my layers to:
```
    layers:
      - F.Cu
      - B.Cu
      - In1.Cu
      - In2.Cu
      - In3.Cu
      - In4.Cu
      - In5.Cu
      - In6.Cu
      - F.SilkS
      - B.SilkS
      - F.Mask
      - B.Mask
      - Edge.Cuts
      - technical
      @EXTRALAYER0@
      @EXTRALAYER1@
      @EXTRALAYER2@
      @EXTRALAYER3@
      @EXTRALAYER4@
      @EXTRALAYER5@
      @EXTRALAYER6@
      @EXTRALAYER7@
      @EXTRALAYER8@
      @EXTRALAYER9@
```
Then I defined the default as empty string in the same file:
```
...
definitions:
  EXTRALAYER0: ""
  EXTRALAYER1: ""
  EXTRALAYER2: ""
  EXTRALAYER3: ""
  EXTRALAYER4: ""
  EXTRALAYER5: ""
  EXTRALAYER6: ""
  EXTRALAYER7: ""
  EXTRALAYER8: ""
  EXTRALAYER9: ""
```

And finally the project can either use this config or if it has its own kibot.yaml then it will be used. And it looks like this:
```
kibot:
  version: 1

import:
  - file: /kibot/kibot-development.yaml
    definitions:
      EXTRALAYER1: '- Stiffener'
```

closed 2023-08-01 by Kedarius

set-soft, 2023-08-01:
Hi @Kedarius !
Note that you should replace all copper layers by `copper`, so your files adapt to N layers.
Additionally, you don't need to specify silk and mask layers because they are already included in `technical`