INTI-CMNB/KiBot #475: [HOW] to implement a filter for Ibom to finish a semi finished PCB

MarkusSBS, 2023-08-02:
Hallo 
What is the recomended way to generate a Bom for hand assemble parts? 

My idea is that I have partnumbers for JLC in my bom and a assembly variant. 
Now i want to generate a ibom with all party with No jlc partnumber but recomended by the Assembly Variant.

Is that easily possible and what funktionality do i have to use for that?

Best regards 

Markus

set-soft, 2023-08-02:
Hallo @MarkusSBS !
I'll suggest creating an iBoM output with a local `dnf_filter` that excludes any component with a JLC part number. Something like:

```yaml
filters:
  - name: not_jlc_parts
    comment: 'Only parts without JLC (LCSC) code'
    type: generic
    invert: true
    include_only:
      - column: _field_lcsc_part
        regex: '^C\d+'
```

This will be applied before the variant, discarding soldered parts, then the variant will take effect discarding parts that aren't for this variant.

set-soft, 2023-10-18:
Closing this issue assuming the answer is correct.