INTI-CMNB/KiBot #476: [FEATURE] Ability to remove(add) solder mask when using variants

Kedarius, 2023-08-04:
I am using variants to create one board file for two variants of MCU. I've create this abomination in KiCad:
[image]
It is basically two footprints overlapping and I have two variants in KiBot: one with small MCU and one with large one. The variants do already remove the solder paste from paste layer which is great. I would like to be also specify that I want to remove the solder mask layer - effectively covering the unused pads with solder mask. 
The reasoning is that I will be ordering few prototypes (of both variants) assembled from JLCPCB and I will need to do it as two separate jobs so I will be using different gerbers anyway - mainly because I do not want to have unused solder paste under the large one. So if the solder mask could be also removed would be great.

set-soft, 2023-08-04:
Hi @Kedarius !

Hmmm ... it look tricky, the solder mask isn't something that simple.
What about adding apertures to the big component? So it removes the solder mask from the exact places where the smaller components are placed. 

To achieve it in a simple way you could add the other 3 footprints to the bigger one. You'll have to work with the pin numbers so they match your design. So you'll get a footprint with the pads for the 4 components. Then you can easily overlap the 3 smaller components with the extra pads in the big one. Not sure if it will work, but is worth trying it.

You could also add a big aperture under the big component, but the smaller ones will be harder to solder.

closed 2023-08-04 by set-soft

Kedarius, 2023-08-04:
Wow, that was fast. I can confirm that it works perfectly:
[image]

And just to clarify, it removes graphics from solder mask layer only?

set-soft, 2023-08-04:
> And just to clarify, it removes graphics from solder mask layer only?

Yes, from the solder mask. This is because you can add arbitrary apertures to footprints, they can be anything you want. In the case of paste you'll never add paste to something that isn't a pad, after all a pad is some copper you want to use to solder something. But with the solder mask is different. Not always 100% correlated to pads.

commit abeea7ec (2023-08-04): [Global options][Add] Option to remove solder mask apertures

`remove_solder_mask_for_dnp` similar to `remove_solder_paste_for_dnp`
but applied to the solder mask apertures.

Closes #476