INTI-CMNB/KiBot #477: [HOW] Blenderoptions with the GitHub Action

Informaticore, 2023-08-04:
**What do you want to achieve?**
I am running the Quick Start variant of this tool (yaml) in my repository but now I would like to add/remove and change some of its options and I have no idea how I can do this. The documentation is just to complex. I found the section for blender but I have no idea how to apply its options to may current workflow file.

**Do you have some PCB/Schematic to use as example?**

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
# SPDX-FileCopyrightText: 2022 nerdyscout <https://github.com/nerdyscout/>
#
# SPDX-License-Identifier: CC0-1.0

name: "KiBot CI"

# Controls when the action will run. Triggers the workflow on push or pull request
# events but only for the master branch
on:
  push:
    paths:
      - '**/*.kicad_sch'
      - '**/*.kicad_pcb'
      - '**/*kibot.yml'
  pull_request:
    paths:
      - '**/*.kicad_sch'
      - '**/*.kicad_pcb'
      - '**/*kibot.yml'
  repository_dispatch:
    types: [run_qs_gha]

# A workflow run is made up of one or more jobs that can run sequentially or in parallel
jobs:
  quick:
    name: "Quick Start (action)"
    runs-on: ubuntu-latest

    steps:
    - name: Get repo
      uses: actions/checkout@v3
      with:
        # So we can run a diff between last 2 changes
        fetch-depth: '0'

    - name: Run KiBot
      # v2_k6 for v1.5.1 has a permission problem
      uses: INTI-CMNB/KiBot@v2_k7
      with:
        quickstart: YES

    - name: Upload results
      uses: actions/upload-artifact@v3
      with:
        name: Automatic_outputs
        path: Generated
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow in GitHub

set-soft, 2023-08-04:
Hi @Informaticore !

There is no magic wand in KiBot, the quick start option is just what its names suggest: a quick mechanism to start using KiBot. Once you start using it is natural that you'll want to customize the outputs, and then is time to invest in reading the documentation. You don't need to read it all, just until you understand how the tool is configured. The docs also contains plenty of examples and links to various complete examples. 

The configuration uses YAML files, and believe me, they are almost closer to English than language than to a programming language. Perhaps you didn't realize that the quick start option generates a configuration file for you and then uses it. Editing the generated file could look challenging at first sight, but once you identify the option you want to customize things become quite simple. Try getting the generated YAML using your artifacts, I think this should work:

```yaml
    - name: Upload results
      uses: actions/upload-artifact@v3
      with:
        name: Automatic_outputs
        path: |
          Generated
          *.yaml
```

Informaticore, 2023-08-05:
@set-soft there is no folder Generated or any YAML file in my artifacts. Where can I find a full example of a YAML file where I might see any or almost every possible option?

Informaticore, 2023-08-05:
@set-soft is there a all-in-one yaml file where I can see all options and configs? The *.yaml option did not work. It generated one yaml file but not much inside

```
multi: 
  VARIANT1:
    site: https://$USER.github.io/$REPO
    summary: 
    readme: README.md
    bom: pcb/docs/bom/*.csv
    eda:
      type: kicad
      pcb: pcb/*.kicad_pcb
    gerbers: gerbers
    color: green
```

set-soft, 2023-08-07:
Hi @Informaticore !
I think you are mixing things:

1. You mention you are using _INTI-CMNB/KiBot@v2_k7_, with quickstart and collecting the stuff using an artifact (_Generated_ dir)
2. You mention stuff related to nerdyscout scripts.

I don't maintan nerdyscout scripts and you should choose one mechanism. 

You can find a reference _all-in-one_ example in the [repo](https://github.com/INTI-CMNB/KiBot/blob/master/docs/samples/generic_plot.kibot.yaml), but the one generated by _quickstart_ is far better.

The YAML you post isn't generated by KiBot.

And for your question: "Is there a yaml collection for the Quick Start option?" the answer is NO, quickstart **generates** a YAML file based on your project and the installed tools, some sort of wizard. And you'll get it generated using the workflow you mentioned, you just need to store it in your artifacts.