INTI-CMNB/KiBot #479: [BUG] Populate tool fails with TypeError

fusedFET, 2023-08-08:
**Bug description**

Running an output of type "Populate" fails with a type error.
A minimal configuration fails, also the one recommended by the documentation.

`src/kibot -c tests/yaml_samples/populate_simple.kibot.yaml -b tests/data/ArduinoLearningKitStarter.kicad_pcb -d example`

I receive following error:
```
- 'Populate example' (Populate) [populate]
Traceback (most recent call last):
  File "src/kibot", line 16, in <module>
    main()
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/__main__.py", line 520, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/kiplot.py", line 527, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/kiplot.py", line 517, in _generate_outputs
    run_output(out, dont_stop)
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/kiplot.py", line 418, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/out_base.py", line 197, in run
    self.options.run(self.expand_filename(output_dir, output))
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/out_populate.py", line 119, in run
    from .PcbDraw.populate import (load_content, get_data_path, read_template, create_renderer, parse_content,
  File "/mnt/c/Users/Guest/Downloads/lnx/KiBot-master/kibot/PcbDraw/populate.py", line 31, in <module>
    def parse_pcbdraw(lexer: Any, m: re.Match[str], state: Any=None) -> Any:
TypeError: 'type' object is not subscriptable
```

**To Reproduce**
You can use the master branch of the repo and run `src/kibot -c tests/yaml_samples/populate_simple.kibot.yaml -b tests/data/ArduinoLearningKitStarter.kicad_pcb -d example`

**Environment**
- WSL2 on windows 10

```
KiBot installation checker

Core:
Linux: 5.10.102.1 (Linux <computername> 5.10.102.1-microsoft-standard-WSL2 #1 SMP Wed Mar 2 00:30:59 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux)
Python: 3.8.10 (default, May 26 2023, 14:05:08)  [GCC 9.4.0]
KiCad: 7.0.6-7.0.6~ubuntu20.04.1
Kibot: 1.6.3

Modules:
Colorama: 0.4.3
LXML: 4.6.4
Lark: 1.1.5
PyYAML: 5.4.1
QRCodeGen: Ok
Requests: 2.22.0
XLSXWriter: 3.0.8
Xvfbwrapper: Ok
markdown2: 2.4.7
mistune: 2.0.5
numpy: 1.24.2

Tools:
Bash: 5.0.17 (GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu))
Blender: 3.6.1 (Blender 3.6.1)
Ghostscript: 9.50.0 (9.50)
Git: 2.25.1 (git version 2.25.1)
ImageMagick: 6.9.10.23 (Version: ImageMagick 6.9.10-23 Q16 x86_64 20190101 https://imagemagick.org)
Interactive HTML BoM: 2.5.0.2 (v2.5.0-2)
KiBoM: Not available
KiCad Automation tools (kiauto): 2.2.6 (pcbnew_do 2.2.6 - Copyright 2018-2023, INTI/Productize SPRL - License: Apache)
KiCad PCB/SCH Diff (kidiff): 2.4.3 (kicad-diff.py 2.4.3 - Copyright 2020-2022, INTI/Salvador E. Tropea - License:)
KiCost: 1.1.18 (KiCost v1.1.18)
KiKit: 1.3.0 (kikit, version 1.3.0)
OpenSCAD: 2019.5.0 (OpenSCAD version 2019.05)
Pandoc: 2.5.0 (pandoc 2.5)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.48.9 (rsvg-convert version 2.48.9)
Xvfb: Ok (xvfb-run)
```

Note: I tried updating mistune to v3, but it seems like a there is a bigger compatibility on V3.

set-soft, 2023-08-08:
> Note: I tried updating mistune to v3, but it seems like a there is a bigger compatibility on V3.

This is a problematic package, I know 0.8.4 works, will check which versions later.

fusedFET, 2023-08-08:
Hey, thanks for the quick update!
Just tried with 0.8.4 but it seems to give the same error.

set-soft, 2023-08-08:
Try removing the `: re.Match[str]` qualifiers. Perhaps Python 3.8.10 doesn't correctly define it.

fusedFET, 2023-08-08:
Good catch!
Removing the [str] actually makes it work.
```
- def parse_pcbdraw(lexer: Any, m: re.Match[str], state: Any=None) -> Any:
+ def parse_pcbdraw(lexer: Any, m: re.Match, state: Any=None) -> Any:

- def parse_pcbdraw(lexer: Any, m: re.Match[str], state: Any=None) -> Any:
+ def parse_pcbdraw(lexer: Any, m: re.Match, state: Any=None) -> Any:
```
I removed it in populate.py.

set-soft, 2023-08-08:
Thanks for testing it. 
I adjusted some details and now the code in the `dev` branch should work even for Python 3.7. 
You don't really need to upgrade.
I also adjusted the _setup.py_ to ask for 3.7 or newer, it was wrongly asking for 3.6 or newer.

closed 2023-08-08 by set-soft