INTI-CMNB/KiBot #480: [BUG] Populate tool interferes with markdown TOC

fusedFET, 2023-08-09:
Hello!

**Describe the bug**

Consider following index.md fed into the 'populate' tool:

```md
# PCBA Manual

- [PCBA Manual](#pcba-manual)
  - [PCB Top assembly](#pcb-top-assembly)
  - [Conclusion](#conclusion)

## PCB Top assembly

First step in the assembly is the populating of the TOP face.

- [[front | ]] This is the front side of the board we are populating

## Conclusion

This is the end of the demo.
```

Note the auto-generated Table Of Contents by vscode below the main title.

**To Reproduce**
Run the populate tool on this index.md.
The rendered results will be:

```md
# PCBA Manual

#### [PCB Top assembly](#pcb-top-assembly)

![step](img/populating_1.png)

#### [Conclusion](#conclusion)

![step](img/populating_2.png)

#### [PCBA Manual](#pcba-manual)

![step](img/populating_3.png)

## PCB Top assembly

First step in the assembly is the populating of the TOP face.

####  This is the front side of the board we are populating

![step](img/populating_4.png)

## Conclusion

This is the end of the demo.
```

The tool seems to see the 'populate' patterns in the TOC. It tries to render unexpected images.

**Expected behavior**

The generated MD to be:

```md
# PCBA Manual

- [PCBA Manual](#pcba-manual)
  - [PCB Top assembly](#pcb-top-assembly)
  - [Conclusion](#conclusion)
 
## PCB Top assembly

First step in the assembly is the populating of the TOP face.

####  This is the front side of the board we are populating

![step](img/populating_1.png)

## Conclusion

This is the end of the demo.
```

**Environment**
Where are you running KiBot:
- WSL2, kibot on dev-branch.

set-soft, 2023-08-15:
Hi @fusedFET !
Did you read the docs?

```
each list item is considered as a single step in populating and will generate an image.
```

And this is exactly what you say you are getting!

set-soft, 2023-08-15:
The above patch adds some basic support for this TOC case. Check if it works for you.

commit 4202f01c (2023-08-15): [Populate][Added] Basic support for regular list items

Closes #480
closed 2023-08-15 by set-soft