INTI-CMNB/KiBot #486: [BUG] `check_pcb_parity` check breaks when symbols have exclude from bill of materials attribute

jtmullen, 2023-08-29:
**Describe the bug**
When symbols have the "Exclude from Bill of Materials" attribute the `check_pcb_parity` check reports Errors. Parts with the attribute are listed as missing from the schematic and all of their connections listed as extra connections, such as: 
```txt
ERROR:J1 found in PCB, but not in schematic (kibot.pre_update_xml - pre_update_xml.py:193)
ERROR:Net `GND` extra PCB connection/s:J1 pin 5 (kibot.pre_update_xml - pre_update_xml.py:193)
```

**To Reproduce**

Check the "Exclude from Bill of Materials" attribute box for a symbol (doesn't have to be a part that this makes sense for) on a design. Run `check_pcb_parity`. It will fail.

**Expected behavior**
I would expect this attribute to not cause this to fail.

 However seeing that way that this is implements I see why it fails. 

**Environment (please complete the following information):**
This bug was noticed because it happens in our CI environment, however I replicated it locally. 

Where are you running KiBot:
- In a CI/CD workflow? Yes: GitHub Actions, using docker container directly, not the github action. 
- Using a docker image? kicad_auto:ki6

jtmullen, 2023-08-29:
Related, I also noticed this in the readme: 
`This is equivalent to the Test for parity between PCB and schematic of the DRC dialog. Only available for KiCad 6.`

Is it correct that this is not available in v7?

We are moving to v7 soon so I may just remove this check now for our system if I have to remove it then anyways.

set-soft, 2023-08-29:
Hi @jtmullen !

> Is it correct that this is not available in v7?

When the docs says something about needing KiCad 6 it means you need KiCad 6 or newer. Most sentences like this were added when KiCad 6 was the last stable version. Of course I can't know if KiCad 7 will support it. In this particular case you need KiCad 6.x or >= 7.0.1, only 7.0.0 is broken.

I'm changing "6" by "6+" in the `update_xml` doc.

I'll investigate your report after finishing some major changes to the docs. I'm moving to the "Read the Docs" format.

set-soft, 2023-08-29:
Hi @jtmullen !

> However seeing that way that this is implements I see why it fails.

Yes, the problem is that this check is part of `update_xml`, which uses the XML generated by KiCad for ... the BoM. So components marked *Exclude from Bill of Materials* aren't saved to the XML file.

But this seems to be a KiCad 6 limitation. In KiCad 7 they get exported, just with a property *exclude_from_bom*.

For KiCad 6 you can use KiBot filters to exclude things from the BoM and avoid using KiCad 6 native mechanisms. Or you can just try KiCad 7.

jtmullen, 2023-08-29:
I see, that is good news. We should be switching pretty soon so that will solve this for us. Thanks!

set-soft, 2023-08-30:
Hi @jtmullen !

The above patch removes the errors. Of course we can't check the parity for the missing components, so you get a warning for them (can be filtered).

I also fixed a couple of misleading errors reported when using KiCad 7. You can see the simple test case I used in the above mentioned commit. 

The included test is checking that KiCad 7 includes the components in the netlist, even when they are excluded from the BoM.

closed 2023-08-30 by set-soft