INTI-CMNB/KiBot #488: kicad7_auto:latest ERROR expanding KICAD6_3DMODEL_DIR

Reid-n0rc, 2023-08-30:
I'm getting an error that `KICAD6_3DMODEL_DIR` cannot be expanded when running kicad7_auto:latest

```
Run make -C CAD/SAO_AMAMW erc
  make -C CAD/SAO_AMAMW erc
  shell: sh -e {0}
make: Entering directory '/__w/SAO_AMAMW/SAO_AMAMW/CAD/SAO_AMAMW'
kibot -e SAO_AMAMW.kicad_sch -b SAO_AMAMW.kicad_pcb -c SAO_AMAMW.kibot.yaml -d Fabrication -m kibot.mak
- Creating makefile `kibot.mak` from `SAO_AMAMW.kibot.yaml`
WARNING:(W008) Unable to find KiCad configuration file (/github/home/.config/kicad/7.0/kicad_common.json) (kibot - config.py:[19](https://github.com/BadgePiratesLLC/SAO_AMAMW/actions/runs/6027109847/job/16351566435#step:4:20)6)
WARNING:(W010) Unable to find KiCad user templates (kibot - config.py:430)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:430)
ERROR:Unable to expand `KICAD6_3DMODEL_DIR` in `${KICAD6_3DMODEL_DIR}/Resistor_SMD.3dshapes/R_080[5](https://github.com/BadgePiratesLLC/SAO_AMAMW/actions/runs/6027109847/job/16351566435#step:4:6)_2012Metric.wrl` (kibot - config.py:112)
Found 3 unique warning/s (3 total)
make -f kibot.mak run_erc
make[1]: Entering directory '/__w/SAO_AMAMW/SAO_AMAMW/CAD/SAO_AMAMW'
- Running the ERC
make[1]: *** [kibot.mak:10[9](https://github.com/BadgePiratesLLC/SAO_AMAMW/actions/runs/6027109847/job/16351566435#step:4:10): Fabrication/SAO_AMAMW-erc_.txt] Error [10](https://github.com/BadgePiratesLLC/SAO_AMAMW/actions/runs/6027109847/job/16351566435#step:4:11)
make[1]: Leaving directory '/__w/SAO_AMAMW/SAO_AMAMW/CAD/SAO_AMAMW'
make: *** [Makefile:[15](https://github.com/BadgePiratesLLC/SAO_AMAMW/actions/runs/6027109847/job/16351566435#step:4:16): erc] Error 2
make: Leaving directory '/__w/SAO_AMAMW/SAO_AMAMW/CAD/SAO_AMAMW'
Error: Process completed with exit code 2
```

set-soft, 2023-08-30:
Hi @Reid-n0rc !

This error isn't fatal, the docker images doesn't contain the 3D models, so you can get such an error.

The command printing the error is:

```
kibot -e SAO_AMAMW.kicad_sch -b SAO_AMAMW.kicad_pcb -c SAO_AMAMW.kibot.yaml -d Fabrication -m kibot.mak
```

And isn't exiting with an error level. From the text you send it looks like the ERC failed when running:

```
make -f kibot.mak run_erc
```

As *BadgePiratesLLC/SAO_AMAMW* seems to be a private project I can't take a look at it. I suggest adding the following environment variable to your workflow: `DEBUG="-vvvv"` 

It will be used by the Makefile to enable debug information. You can also store the artifacts even on error and take a look at the reports. The `Fabrication/SAO_AMAMW-erc_.txt` should contain the ERC errors. The error level is *ERC_ERROR* (10).

Reid-n0rc, 2023-08-30:
Thank you for the quick reply @set-soft!

`Fabrication/SAO_AMAMW-erc_.txt` looks like it's not being created because Retrieve results is being skipped.
[image]

Reid-n0rc, 2023-08-30:
I've moved past this error but I cannot get it to generate output in the fabrication folder. 

I've created a public project that is very simple to use to troubleshoot. [https://github.com/Reid-n0rc/ci-cd-test](https://github.com/Reid-n0rc/ci-cd-test)

set-soft, 2023-08-31:
> `Fabrication/SAO_AMAMW-erc_.txt` looks like it's not being created because Retrieve results is being skipped.

Please read the GitHub docs. You can mark a step with: `if: ${{ always() }}` to be run even if the previous step failed.

set-soft, 2023-09-05:
Hi @Reid-n0rc !

Looking at your repo it looks like the problem was a wrong path.

Note that you no longer need to include the manufacturer templates. Now they are internal templates that you can import and customize.

closed 2023-09-05 by set-soft